cjcliffe/CubicSDR #664: Couln't create Opengl context

ClaudeMa, 2018-07-04:
hi 

I have build Cubicsdr on 3 linux platform Debian 9, Ubuntu 18.04 and Fedora 97
followings instructions here
https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux 
because i need sdrplay RSP1A support.
On all platform, when i start CubicSdr i receive "Couln't create Opengl context".
If i download an appimage when starting, no message but sdrplay not detected.
Is there something wrong or missed in the wiki page?
thanks
Claude

MrJBSwe, 2018-07-26:
I have the same problem on Ubuntu studio 18.04
( but CubicSDR will run after the error, without opengl ? )

observations
 opengl samples for wxwidgets   cube,penguin etc  -  works 
 CubicSDR-0.2.4-x86_64.AppImage - work without error 
 running without nvidia drivers work ( without opengl I guess ..slow )
 Ogre3D demos works

cjcliffe, 2018-08-04:
@ClaudeMa @MrJBSwe I think this is related to wxWidgets 3.1.0; I'll update the instructions but basically just replace wxWidgets-3.1.0 with wxWidgets-3.1.1 in the guide.

ClaudeMa, 2018-08-11:
Many Thanks for reply Charles
Now work with wxWidget 3.1.1
No error message and button are now "readable"

closed 2018-08-11 by ClaudeMa