cjcliffe/CubicSDR #665: Bookmark label edit fail

j3ngel, 2018-07-13:
[OS: macOS High Sierra]

I can create a bookmark group and add items there by dragging them from the "Active" section. 

However, in the bookmark section, I cannot change the label. The empty label input box is there and I can type in it, but, when I hit Enter I get an alert sound and nothing else in terms of feedback why it won't save/accept the change.

cjcliffe, 2018-08-04:
@reallyallnamestaken ahoy, thanks for the report -- I'll be upgrading to HS shortly so I'll take a look at the cause for this.

npaisnel, 2018-11-04:
As a work around as this is still here in 0.2.5, I use the File|Bookmarks 'Save Bookmarks As" function.

Then you have to go to the created xml file and edit the section for the book mark adding the name yu want between the user label tags

<user_label>ATIS</user_label>

christianallen808, 2019-03-18:
Hey Charles,

I have this issue as well.  No problem creating groups, just editing each freqs with "label" bar.  I've used previous versions of CubicSDR and it used to work flawlessly.  I had to reinstall CubicSDR on this MacBook and am having problems.

Currently running CubicSDR (v.0.2.4) on macOS Mojave (10.14.3)

Thank you for what you do.  Love CubicSDR!

BrotherPhil, 2020-05-10:
I also have this issue, on El Capitan; I've tried editing the labels, but get a forced close when I try to load the edited file.

Observerist, 2020-06-11:
Any news about a possible update that would enable the bookmark labels ?

Has there been any developments for the release of a new version or is everything halted around Cubic ? Using the app on a Mac under Mojave. 
Just hope this is not the end ....

Cheers & Stay Safe

cjcliffe, 2020-06-17:
Making bookmarks a priority at the moment; seeing a few different issues that appear to be related to wxWidgets updates -- and not just on macOS.

cjcliffe, 2026-11-14:
Several fixes related to bookmarks and label editing including this one have been fixed in master branch

closed 2026-11-14 by cjcliffe