cjcliffe/CubicSDR #666: CubicSDR start aborts with wxGL error message

Rallyefahrer, 2018-07-14:
After starting CubicSDR, it aborts with following wxGL error message: 
ASSERT INFO:
../src/unix/glx11.cpp(158): assert "fbc" failed in wxGLContext(): invalid GLXFBConfig for OpenGL

BACKTRACE:
[1] wxGLContext::wxGLContext(wxGLCanvas*, wxGLContext const*)
[2] wxEntry(int&, wchar_t**)
[3] __libc_start_main
[4] _start

Rallyefahrer, 2018-08-04:
Hey, I'm using the official debian package:
https://packages.debian.org/sid/cubicsdr
and the NVIDIA propr. driver:
https://packages.debian.org/sid/nvidia-legacy-340xx-kernel-dkms

pe1chl, 2018-12-26:
I see the same issue on exactly the same OS/hardware combination (nvidia proprietary driver).
Is there any workaround without de-installing the nvidia driver?

Rallyefahrer, 2018-12-29:
> I see the same issue on exactly the same OS/hardware combination (nvidia proprietary driver).
> Is there any workaround without de-installing the nvidia driver?
Maybe it has something to do with the big X-server-update to 1.20 in combination with nvidia propr. driver. The issue occured at at same time as far as I remember.

pe1chl, 2018-12-29:
I will try on a different system with Debian buster/sid but with intel video, but it will be next year.
I intended to do so last week (at work) but I forgot.

pe1chl, 2019-01-02:
Tried it, and the same issue arises.  So it is not directly related to the Nvidia driver.
(this system has an Intel G45 mobile graphics adapter, driver is i915)

Rallyefahrer, 2019-01-03:
Then it is maybe related to X-server 1.20, which conflicts somehow with CubicSDR and / or wxGL

pe1chl, 2019-01-03:
Yes that could be.  I only wanted to add this info because I have the same legacy-nvidia-340xx driver in use and I feared it might be related to that (or otherwise more people would be affected).
But probably not so, it would likely not work at all on Debian buster/sid.  Unless we find someone for whom it works OK.

Rallyefahrer, 2019-01-03:
Yeah, thats fine. But the problem started only after begin of July 18 - while before everything worked fine -  this must be related to one of the updated packages in Debian Sid in this time. And one of the major updates in this time was the X-server from 1.19 to 1.20, but for sure other relevant packages were also updated.

Rallyefahrer, 2020-03-08:
Update: with Debian package cubicsdr 0.2.5+dfsg-3 on testing / Bullseye, the problem is gone. Issue could be closed from my point of view !

closed 2020-03-08 by Rallyefahrer