cjcliffe/CubicSDR #667: CubicSDR-0.2.4-win64: no rtl-sdr support.

hhrhhr, 2018-07-27:
CubicSDR 0.2.4 win64 builded with VC++2015, but rtlsdr.dll — with VC++2013.
As the bundle contains only "Microsoft Visual C++ 2015 Redistributable (x64) - 14.0.24215" (vc_redist\vc_redist.x64.exe), then there is no support for rtl-sdr.

Solution: include vcr2013 in the package or recompile rtlsdr.dll with a newer compiler.

cjcliffe, 2018-08-04:
@hhrhhr I don't think I've had issues with it before; is there an error you're getting related to missing vc 2013 or the device just isn't showing?  I suppose I can just re-compile it with 2015+ so that the 2013 package isn't needed.

hhrhhr, 2018-08-04:
If VC2013 is missing then no rtllsdr devices apepairs in select menu. Compiling all components within one version of the compiler would be the best idea.

vsonnier, 2018-08-04:
@hhrhhr @cjcliffe FYI I use https://github.com/librtlsdr/librtlsdr compiled with  VC++2017 (2015 should be fine too) and it works fine for either 64 or 32 bit CubicSDR.

Maitaikawika, 2018-08-21:
In case anyone else happens upon this thread, I got it working with my rtl-sdr by installing Microsoft Visual C++ 2010 Redistributable and Microsoft Visual C++ 2013 Redistributable onto my computers, reinstalling the WinUSB driver in Zandig, and then relaunching CubicSDR. There may be more elegant ways to do it, but this workaround worked for me.

XE3ADC, 2020-08-07:
Many Thanks Maitaikawika, my CubicSDR(V2.4) installation on w7-64 would not show a rtl-sdr usb connected device on the sdr device chooser window, installing the VC++ 2013 resolved the problem, although in my case it was not nessesary to install VC++ 2010 neither do the zadig procedure. However, as hhrhhr pointed above, if CubicSDR installer for Win64 contained the VC++ 2013 redistributable It would became a nice rounded package
In the meantime, It would also be nice to include a note on the CubicSDR download link informing users about this issue, perhasp with a little procedure for the fix.
Thanks all for the efforts...
Cheers!!

vsonnier, 2022-01-15:
Outdated issue. Closing.

closed 2022-01-15 by vsonnier