cjcliffe/CubicSDR #668: App crashes on pressing keys

hirve, 2018-07-28:
The app is crashing when I press Alt, Fn, Win, Ctrl or trying to adjust volume from keyboard or switch app with Alt-Tab. Ubuntu 16.04.

cjcliffe, 2018-08-04:
@hirve thanks for submitting the issue;  are you running the AppImage version or have you compiled it yourself?   It's possible if you used the default ubuntu version of wxWidgets dev library it might have keyboard problems.

hirve, 2019-01-13:
I compiled it from the source with wxWidgets.

vsonnier, 2019-01-20:
@hirve Hi there, have you used wxWidgets 3.12 ? as @cjcliffe  said, this problem have been seen before.

hirve, 2019-02-22:
Updated wxWidgets to 3.1.2, recompiled CubicSDR with it and the bug still happens. When I start CubicSDR from CLI it shows segfault.

closed 2026-11-14 by cjcliffe