cjcliffe/CubicSDR #670: Use system-provided rs232, tinyxml when possible

mdomsch, 2018-08-07:
Just you have built CubicSDR to use system-provided libraries whenever possible, can you please provide a way to discover and use system-provided librs232 and tinyxml when available?  These are provided in Fedora packages librs232-devel and tinyxml-devel.  Fedora cannot ship packages that include third party libraries statically built into the application when they are available in the distribution already.

vsonnier, 2018-08-07:
@mdomsch All these external libs are almost "single file" dependencies, precisely easy enough to integrate in the source code. TinyXML is the latest version and will stay so because developpement has shifted over TinyXML2. 
Another advantage is to allow to do some small fixes on the code, such as in [MSVC serial fix-up](https://github.com/cjcliffe/CubicSDR/commit/21b2b3d1917ecdf103fd0b4c778c1b955c1342d5).

IMHO providing them externally will only create problems. 
Maybe @cjcliffe has another POV to make, so I'll leave the issue open.

mdomsch, 2018-08-07:
It's not exactly forbidden, but it is strongly discouraged by both Fedora and Debian.

https://fedoraproject.org/wiki/Bundled_Libraries?rd=Packaging:Bundled_Libraries
https://www.debian.org/doc/debian-policy/ch-source.html#s-embeddedfiles

I can explicitly mark the bundling in the rpm spec file if necessary.

vsonnier, 2018-08-08:
Right, these are pretty usual recomendations for statically compiled code in general: could be not up-to date, could be buggy and can't be fixed...and so on.
In our case, none of these advices apply and the praticallity of having a definite version for the source-included library, some with small customization, beats the purity of "static code is bad".

Since the inclusion is "by source", I don't think it can be considered as bundled dependencies anyway. 

@cjcliffe has purposefully direclty added those libs for no-nonsense compiling and saving the trouble of requesting dependencies for such small libs. 
By comparison we always have often questions poping up, for every system imaginable, on how to get the right version and compile SopaySDR, wxWidgets, and the other big dependencies.

At least including those small libs as source save us that kind of support.

mdomsch, 2018-08-08:
Understood.  I've noted this issue in the spec file so anyone who goes looking will find it, and have added the following Provides:
`
bundled(librs232) = 0.21
bundled(tinyxml) = 2.6.2
`

closed 2018-08-08 by mdomsch

vsonnier, 2018-08-08:
If you are at that, maybe https://github.com/cjcliffe/CubicSDR/tree/master/external/lodepng should be listed too :) . 
Anyway, have fun with CubicSDR !

mdomsch, 2018-08-08:
We don't already have loadpng in Fedora, so there's nothing to note at this
time.  Thanks for the software.