cjcliffe/CubicSDR #679: RfNotch and DabNotch

Toontje, 2018-09-07:
I just received my RSP1A and one of the differences i see compared to my RSP1 i had before is that now i have three new settings: BiasT, RfNotch and DabNotch.
I know what is BiasT, but what is RfNotch and DabNotch and how do i use them in CubicSDR?

Toontje, 2018-09-07:
Forget it. I found it here: https://www.youtube.com/watch?v=qcDe6swBNwg

vsonnier, 2018-09-15:
Hello @Toontje ! Well I see you figured out for yourself those settings :) As far as I know, all the available RSPs settings are supported in CubicSDR (by SopaySDRPlay) including as you've seen the RF, DAB notch filters even if the name of the settings are not immediatly obvious.

Comparing to SDRUno, the (only?) thing not supported in SoapySDRPlay is the decimation control, but it can't be since the SoapySDR exposed functions do not include such possibility. Instead, the SoapySDRPlay driver choose the "best" decimation according to the selected sample rate.

closed 2018-09-15 by vsonnier