cjcliffe/CubicSDR #681: Peak hold has no effect on the waterfall.

Magalex2x14, 2018-09-13:
It seems like enabling Peak hold function has no effect on the waterfall...

[image]

The floating gain level of the waterfall gives the wrong impression of a changing noise level with enabled Peak hold. This is often annoying :)

Magalex2x14, 2018-09-13:
Ooops... Some more additions required:

AppFrame.cpp 
after 
_2960 wxGetApp().getSpectrumProcessor()->setPeakHold(!wxGetApp().getSpectrumProcessor()->getPeakHold());_
add
`wxGetApp().getAppFrame()->getWaterfallDataThread()->getProcessor()->setPeakHold(!wxGetApp().getSpectrumProcessor()->getPeakHold());`

CubicSDR.cpp
after 
_630 getSpectrumProcessor()->setPeakHold(getSpectrumProcessor()->getPeakHold());_
add
`getAppFrame()->getWaterfallDataThread()->getProcessor()->setPeakHold(getSpectrumProcessor()->getPeakHold());`

SpectrumCanvas.cpp
after
_311 wxGetApp().getSpectrumProcessor()->setPeakHold(wxGetApp().getSpectrumProcessor()->getPeakHold());_
add
`wxGetApp().getAppFrame()->getWaterfallDataThread()->getProcessor()->setPeakHold(wxGetApp().getSpectrumProcessor()->getPeakHold());
`

[waterfall_PeakHold.diff.txt](https://github.com/cjcliffe/CubicSDR/files/2380785/waterfall_PeakHold.diff.txt)

vsonnier, 2018-09-15:
Hello @Magalex2x14. Well, I never suspected peak hold functions would be available for the waterfall display in the first place. 
So basically the Waterfall keep moving, but the colors get frozen... but I don't know if should be applied 
systematically on __both__ Spectrum and Waterfall.
I would suggest something like :
- 1) The button 'P' would indeed activate/deactivate peak hold on __both__ Spectrum and Waterfall,
- 2) While the shortcut 'P' would be a 3 circling state PeakHold Spectrum / PeakHold Spectrum+Waterfall / Off.

Personally I only use the shortcut, so the button is just a visual clue.

vsonnier, 2018-09-15:
So finally you agree partially about me about having the Peak Hold not be intrusive, so leaving the waterfall unchanged. One thing that could be a complement is a "min peak hold", suggested in #532. 
Closing this issue then ?

Magalex2x14, 2018-09-15:
No, i think that PeakHold must affect both the Spectrum and the Waterfall. 
But peak hold must be always disabled for Demodulator spectrum and waterfall.

closed 2026-11-14 by cjcliffe