cjcliffe/CubicSDR #682: AppImage on Linux missing Ma

HeroCC, 2018-09-22:
I'm trying to get CubicSDR to work on my desktop, but I am having some issues. Very few elements show up, and I do not get any audio output. https://i.imgur.com/7yV0gRk.png. I am using an RTL-SDR USB stick, and I `modprobe -r dvb_usb_rtl28xxu` before launch. 

The console gets spammed with this: 
`(CubicSDR-0.2.5-x86_64.AppImage:24067): GLib-CRITICAL **: THE_TIME: g_error_free: assertion 'error != NULL' failed` whenever I move my mouse over an element, and at the beginning this error is printed:
```
[ERROR] SoapySDR::loadModule(./////////lib/SoapySDR/modules0.7/libsdrPlaySupport.so)
dlopen() failed: libmirsdrapi-rsp.so.2.13: cannot open shared object file: No such file or directory
```

Any help you can give me would be much appreciated, thank you!

scopelemanuele, 2018-10-13:
I have the some problem in manjaro linux.

NotAFile, 2018-10-16:
I have the same issues, but I don't have any of the console output you noted.

vsonnier, 2018-10-17:
@HeroCC Hello! 
Looks like you want to use a RTL-SDR. Unfortunatly, your screenshot shows that you don't have selected the device *on the left pane* of the RTL-SDR device, so it cannot work that way.

The `libsdrPlaySupport.so` log means it wants to open the SDRPlay module. If you don't use this kind of device, you can ignore the log I guess.

@NotAFile @HeroCC Please describe your Linux system in terms of distro, SDR drivers installed, ...etc. and also attach the CubicSDR console log.

cjcliffe, 2019-01-03:
Hmm, even if the device didn't work the missing elements are OpenGL specific; which leads be to believe it's a OpenGL usage or graphics driver issue.  Will investigate.

Chryseus, 2019-03-10:
It's caused by outdated wxWidgets, building it for 3.1.2 on Arch fixed it for me.

closed 2026-11-14 by cjcliffe