cjcliffe/CubicSDR #683: CubicSDR CubicSDR-0.2.4 in not work

Abdula-2015, 2018-10-10:
I used **CubicSDR-0.2.4-x86_64.AppImage**. **Debian 9/32 bit** and dongl **RTL2832U+R820T**.
CubicSDR starts, but does not work.
No reception, no waterfall.

But Gqrx works great!

Why does not work CubicSDR ???

Osirus126, 2018-10-10:
I have a feeling you are using the rtl_tcp application for a SDR server but you must use SoapySDRServer for CubicSDR.

NotAFile, 2018-10-16:
I have the same issue. CubicSDR from my repos works fine, but the AppImage shows no output.

While it's running, it shows the following output:

```
SDRThread::readStream(): 3.1 iqDataOutQueue output queue is full, discard processing of the batch...
SDRThread::readStream(): 2. SoapySDR read failed with code: -1
```

Abdula-2015, 2018-10-16:
I do not understand why CubicSDR was made.
I use Gqrx - it works!

CubicSDR creates fatal problems, but developers are lazy and do not help.

Qqrx is **better**! And I return to it.

NotAFile, 2018-10-16:
@Abdula-2015 You're being extremely rude here. Cjcliffe has put a lot of free time into making this and making it available for free. That is anything but lazy. People like you complaining and insulting them is a reason why so many developers feel frustrated and give up on great projects.

vsonnier, 2018-10-17:
Dear @Abdula-2015, since your are *not* lazy as the developers here, as shows your commit history (*sarcarsm*) you can always compile the whole thing from scratch as the Wiki and multiple issues already describes. 
As for Applmage in general, it just a convenience that *may* work if all the dependencies are satisfied on the running machine. And the correct drivers and libraries installed. 

Finally, since we are lazy but still not telepaths, we cannot guess the multitude of Linux variants out there  so the condition to be helped is to describe your system kernel, distro, *SDR drivers*, packages, SDR device you want to listen, extensive logs of Cubic ...etc.

Otherwise on Linux, the statement  "Doesn't work on my [random Linux distro name here] is just, well, spam.

Please continue on #682 providing the above info as described, for those still interested.

closed 2018-10-17 by vsonnier