cjcliffe/CubicSDR #684: Feedback on why rig control fails

Toontje, 2018-10-17:
I am currently struggling to get rig control working. But the ticket is not about this. In this ticket i would like CubicSDR to provide feedback on why rig control fails. Currently the only thing you notice is that Enable Rig is unchecked. If you perceive that all settings are correct and you cannot get Enable Rig to remain checked, CubicSDR should provide feedback on why Enable Rig gets unchecked by CubicSDR.

cjcliffe, 2019-01-03:
@Toontje agreed; there's really no indication of anything for hamlib at the moment and there should be.

cjcliffe, 2020-07-14:
Dialog will now show text for all hamlib errors when rig communication fails in addition to un-checking the menu.

closed 2020-07-14 by cjcliffe