cjcliffe/CubicSDR #685: Dark mode in Mojave

Toontje, 2018-10-17:
Please support dark mode in Mojave.

cjcliffe, 2019-01-03:
@Toontje will do what I can -- unfortunately I don't have any Macs here that can run Mojave or the XCode released for it at the moment.

Magalex2x14, 2019-01-13:
Seems like dark mode in Mojave works pretty well with wxWidgets 3.1.2... I do not see any problems with this version (just replaced version number in OS X build instructions from [wiki](https://github.com/cjcliffe/CubicSDR/wiki/Build-OSX), but I cannot check how it works with previous releases of OS X. I came to this when ran into a problem trying to build CubicSDR in Mojave (10.14.2) with wxWidgets 3.1.1 - dynamic interface elements were not displayed (meters, waterfalls, spectrums, as long as i remember). Compiling with wxWidgets 3.1.2 solved the problem (plus dark mode as a bonus).

vsonnier, 2019-01-20:
Hello @Magalex2x14, I wish you a happy new year !

Since dark mode is working for you, could you please post some screenhots of it ? 
If then their look are right according to @Toontje and other users, we may consider the issue resolved.

It would be then only a matter of changing the MacOS Wiki instructions to point on wxWidgets v3.12.

Magalex2x14, 2019-01-20:
@vsonnier Sure! The dark red color of the peak line is my only change, made quite a long time ago (            glColor4f(1.0, 0.0, 0.0, 0.15)).

[image]
[image]

vsonnier, 2019-01-23:
Thanks @Magalex2x14, it looks like the Dark mode is indeed working on the non-OpenGL elements as expected, with inverted colors. The rest of OpenGL is already Dark :) of course.

@Toontje Do you think the issue could be closed ? I've already updated the Wiki to reference v3.12 when building on OSX. (same as for Linux instructions)

Toontje, 2019-02-10:
Yep! Close it! Thanks!!

vsonnier, 2019-02-12:
Thank you both !

closed 2019-02-12 by vsonnier