cjcliffe/CubicSDR #687: Wishlist??

joel-1, 2018-10-31:
I just started using CubicSDR and like the way things are displayed. I want to use this as a panadapter for my radio and not an SDR receiver. It connects to my rig just fine and tracks frequency, sets frequency. What I notice,

When I switch bands from the radio, the demodulator spectrum and waterfall update but not the main spectrum or waterfall.

Pressing mode buttons on the radio does nothing, selecting demodulator type does not update the radio. I do not find functions for this in the code so not a surprise.

My panadapter wish list would be,

Changing bands via the radio updates the demodulator ( seems to do that ) and the main display. The main display automatically centers and goes to the full span for the band.

Mode buttons work both ways.

Demodulator span follows radios filter width setting from radio.

My angle is HAM radio centric. I am looking at the code to see how to implement these things. For now it's a lot to look through and understand what all the knobs are and where.

Be nice to have a "HAM" mode to set these features.

Thanks for all the great efforts!

vsonnier, 2018-11-01:
@joel-1 What you want already exist under the menu "Rig control". See this video
[https://www.youtube.com/watch?v=dKCRDLz0cI8](Setting Up a Panadapter on your Mac with the RSP1A and CubicSDR)
as an example using a RSP1a (but it should work with any device) and a Mac.

joel-1, 2018-11-01:
Hi,
I did see this video and things do not work as I describe. Mode changes from the radio do not reflect into Cubic. When I change bands, it only changes the demodulator spectrum. The main spectrum display does not change. The only way to work is to interact with the Cubic GUI. I do have the rig control items set up and it tracks the VFO fine, just nothing else.  I believe hamlib does support my radio fine as other SDR software does work.

Operating from the GUI is as expected, but not from the radio.

colinluthier, 2021-11-27:
@cjcliffe This is a great start on a panadapter. I have been using SDR Console on my Windows but but I am trying to move it to a Linux box. Having the mode buttons work both ways would be an amazing step forward. I will be looking at a band and switch modes rapidly as I am looking for signals. I there is anything I can do to help please let me know. I am by no means an object oriented guru. I did find .set_mode and .get_mode in the  hamlib files so I know that part is done.