cjcliffe/CubicSDR #688: Cubic Crashes when Starting a Locla network RSP1a on a RPi

npaisnel, 2018-11-01:
It worked fine yesterday, and with no changes, today Cubic refuse to start.

Tried 0.2.3 and 0.2.4
 I saw a previous issue with Cubic crashing and a DMG to won load with 0.2.2 which gave a log file in a terminal that it opened..no idea if this is of any help..but to start with I will let you see this . and then I can fill you in on anything else.
OSX 10.11.6 on mac Pro Twin Quad Core Zeon
The Pi that the RSP1a is connected to is a 3B+ running Raspbian lite 
Running SoapyRemote.

It worked fine yesterday..today no joy.  have rebooted both devices multiple times now.

[CubirRSP.txt](https://github.com/cjcliffe/CubicSDR/files/2538149/CubirRSP.txt)

npaisnel, 2018-11-01:
This is when it works. 

I changed the sample rate before i hit 'Start ' this time, to 500hz, but that is not it.  I can quit out and go back in again. and start it on different sample rates and it works.

Seems quite random as to if it crashes or not.

Pi is on a local network connected on WiFi.

[Cubic 024.txt](https://github.com/cjcliffe/CubicSDR/files/2538192/Cubic.024.txt)

vsonnier, 2018-11-01:
@npaisnel Hello, looks like you are runing an incredibly old Cubic version; __0.1.2x__,  __NOT__ 0.24. 
Besides, the most recent is 0.25, and can be found [here](https://github.com/cjcliffe/CubicSDR/releases).
Please try the most recent version.

npaisnel, 2018-11-01:
Ah
As I said, the first time I ran it from terminal I ran it from a link from another thread on here, I thought it maybe was a diagnostic version that allowed it to run from terminal

If you look at my second out put from terminal text file, I worked out how to run my current version from the command line by using ‘Show package content’ and dragging the actual executable from the package to command line

The crash did occur in both version 0.2.3 and 0.2.4 though,

I will try 0.2.5 from your link and get back to you

npaisnel, 2018-11-02:
OK, stil crashing with 0.23 0.24 and 0.2.5

here is text output from running in terminal

[Cubic025 Soapy Crash.txt](https://github.com/cjcliffe/CubicSDR/files/2541764/Cubic025.Soapy.Crash.txt)

SDRplay, 2018-11-02:
What build of SoapySDRPlay do you have on the Pi? Are you running the image that we provide? Just want to understand if there's any compatibility issues.

npaisnel, 2018-11-02:
Sorry I am unable to answer your question about which build of SoapySDRPlay I have, it is which ever is downloaded from setting up using that tutorial I linked to further up.

If you do provide a headless Pi image with auto boot of SoapyRemote, then I'll happily try it

npaisnel, 2018-11-02:
Umm..with my Pi connected via ethernet cable, I can now SSH to it to shut it down either via its ethernet IP address, or its WiFi IP address....  So effectively it is now connected to the router via two IP's...

Does that have a bearing?

If so how do i disable/ re enable WiFi on th Pi via SSH?

My ultimate aim is for ethernet connection, headless system, bound to specific IP address and port auto booting Soapy Remote on Pi reboot, and port forwarding to enable connection while away from home.

SDRplay, 2018-11-02:
I use Odroids personally now as they are true Gigabit ethernet controllers. On a LAN I can get 10 MHz sample rate using SoapyRemote. I have built 2 headless server images...

https://www.sdrplay.com/software/SDRplay_OdroidC2_Headless_V0.1.img.7z

https://www.sdrplay.com/software/SDRplay_OdroidXU4_Headless_V0.1.img.7z

The RPi image we provide also works, but SoapyRemote needs to be started manually (or through the menu system we startup), but you can easily make SoapyRemote start up on boot up. I would keep the sample rate low on RPi, due to the way all I/O goes through the USB stack.

Hope that helps,

Best regards,

Andy

npaisnel, 2018-11-02:
Oh, OK, so you are part of the SDRPLAY ‘people ‘. Did not get the connection

Running this headless in the attic , 
Did start with the SDRPlay image but was running it headless so did not see the need for GUI based image.

Will you likely be producing a RPi Lite Image  for remote use ?
I had conversation with SDRPlay Support Helpdesk Team under, 
iPhone app ? [#594432]

And they did suggest asking you to see about creating a Pi headless image

Yes I can bind to a port using. ‘- - bind ‘command 
But that does not survive the reboot

Guessing the answer is to put IP address and port in the systemd service file ?   But I’d have no idea how to do that

Odroid.... umm another device to buy... i’ll look them up.  So you are saying and Odroid will stream the full 10Mhz of the RSP1A?   Interesting might have to look

But in reality all I am doing this for is to monitor single frequency ( airfield weather) while away from home base airfield

npaisnel, 2018-11-02:
I like the Odroid devices..but I only need to monitor one frequency so the Pi will do me for now

Can you show me how to get Soapy Remote to start automatically from boot with a specified port 
Got it auto starting with systemd service at moment but can specify the port

So I can port forward to it, and access it from outside my network

250Khz is fine as I only want to monitor single frequency

SDRplay, 2018-11-02:
I suggest closing this issue and raising one on SoapyRemote issue list as Josh will most likely know. I'll start an issue over there and include you.

npaisnel, 2018-11-02:
Cheers

vsonnier, 2019-03-31:
Time to kill this stale issue.

closed 2019-03-31 by vsonnier

Thinh1997, 2023-04-28:
> I was just about to post his when you replied Not sure which version..I built a headless system from this tutorial... https://ansi.23-5.eu/2018/01/sdrplay-remote-raspberry-pi/
> 
> starting Soapy Remote at boot using systemd and systemctrl enable SoapySDRServer
> 
> OK, looks like it is a sample rate issue combined with a network issue
> 
> Once running, after starting at 500khz, if I change to 2.048Mhz sample rate from 500kHz it crashes..see terminal text attached
> 
> 1Mhz it works. 2.00 mhz and above Cubic crashes out
> 
> This is with the Pi connected via WiFi hardwired to ethernet cable it it starts up to 3mhz.
> 
> Changine to above that in Cubic and it crashes out again even when hard wired

This is what exactly I'm dealing with. I have a USRP B200 and B210. Both of them make the CubicSDR 0.2.4 (I'm running CubicSDR in folder bin of PothosSDR) crash when choosing bandwidth 16MHz. If you choose bandwidth 2.66666MHz, everything will be fine. I'm still stuck it this issues, still have no idea where it comes from. 

My Visual Studio telling me that it has error in allocate memory with Boost library in function runDemodChannels(int channelBandwidth) of SDRPostThread.cpp. That's all I know. I'm still dealing with it in hopeless, hope that you can give me some advice.