cjcliffe/CubicSDR #691: Build failure

parkerlreed, 2018-12-04:
soapysdr-git 3:0.7.0.r5.g7716dfb-1
liquid-dsp-git v1.3.1.r232.gf7b9928c-1
libicns 0.8.1-5
wxgtk3-dev 3.1.1-1
gcc 8.2.1+20180831-1

````
[ 95%] Building CXX object CMakeFiles/CubicSDR.dir/external/cubicvr2/math/cubic_math.cpp.o
[ 96%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SoapySDRThread.cpp.o
[ 97%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestCanvas.cpp.o
[ 98%] Building CXX object CMakeFiles/CubicSDR.dir/src/ui/UITestContext.cpp.o
[100%] Linking CXX executable x64/CubicSDR
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/FrequencyDialog.cpp.o: in function `std::_Rb_tree_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range> > std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range> > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range> >, std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>&&, std::tuple<>&&)':
FrequencyDialog.cpp:(.text._ZNSt8_Rb_treeINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4pairIKS5_N8SoapySDR5RangeEESt10_Select1stISA_ESt4lessIS5_ESaISA_EE22_M_emplace_hint_uniqueIJRKSt21piecewise_construct_tSt5tupleIJRS7_EESL_IJEEEEESt17_Rb_tree_iteratorISA_ESt23_Rb_tree_const_iteratorISA_EDpOT_[_ZNSt8_Rb_treeINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4pairIKS5_N8SoapySDR5RangeEESt10_Select1stISA_ESt4lessIS5_ESaISA_EE22_M_emplace_hint_uniqueIJRKSt21piecewise_construct_tSt5tupleIJRS7_EESL_IJEEEEESt17_Rb_tree_iteratorISA_ESt23_Rb_tree_const_iteratorISA_EDpOT_]+0x91): undefined reference to `SoapySDR::Range::Range()'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::setSoapyDevice(SoapySDR::Device*)':
SDRDeviceInfo.cpp:(.text+0x79a): undefined reference to `SoapySDR::Device::unmake(SoapySDR::Device*)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getSoapyDevice()':
SDRDeviceInfo.cpp:(.text+0x7cd): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::hasCORR(int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0x9f9): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getAntennaNames[abi:cxx11](int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0xa82): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getAntennaName[abi:cxx11](int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0xb27): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getCurrentGain(int, unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
SDRDeviceInfo.cpp:(.text+0xdac): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::~SDRDeviceInfo()':
SDRDeviceInfo.cpp:(.text+0xf3b): undefined reference to `SoapySDR::Device::unmake(SoapySDR::Device*)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getSampleRates(int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0x1807): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getGains[abi:cxx11](int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0x1ece): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDREnumerator.cpp.o: in function `SDREnumerator::enumerate_devices(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool)':
SDREnumerator.cpp:(.text+0x28ef): undefined reference to `SoapySDR::getAPIVersion[abi:cxx11]()'
/usr/bin/ld: SDREnumerator.cpp:(.text+0x2989): undefined reference to `SoapySDR::getABIVersion[abi:cxx11]()'
/usr/bin/ld: SDREnumerator.cpp:(.text+0x2a23): undefined reference to `SoapySDR::getRootPath[abi:cxx11]()'
/usr/bin/ld: SDREnumerator.cpp:(.text+0x2b95): undefined reference to `SoapySDR::listModules(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
[... cut ...] oapySDR7ArgInfoEESt10_Select1stISA_ESt4lessIS5_ESaISA_EE22_M_emplace_hint_uniqueIJRKSt21piecewise_construct_tSt5tupleIJRS7_EESL_IJEEEEESt17_Rb_tree_iteratorISA_ESt23_Rb_tree_const_iteratorISA_EDpOT_]+0x50): undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1332: x64/CubicSDR] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
````

guruofquality, 2018-12-06:
I had a few bugs fixes for the cmake work since the merge. And while I was at it, I decided to set those variables as well, I guess its easier that way. So If you pull and rebuild SoapySDR (if you are using master), this should be fine again.

parkerlreed, 2018-12-06:
Cheers. That does work.

closed 2018-12-06 by parkerlreed