cjcliffe/CubicSDR #693: can not compile on ubuntu mate 18.10

jimmycapizzi, 2018-12-05:
done it many times on other machines. i followed the wiki for the build instructions exactly, and this error always get, i am stumped here fellas.
 --------------------------------------------------------------------
[  1%] Linking CXX executable x64/CubicSDR
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/FrequencyDialog.cpp.o: in function `std::_Rb_tree_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range> > std::_Rb_tree<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range>, std::_Select1st<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range> > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, SoapySDR::Range> >, std::piecewise_construct_t const&, std::tuple<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&>&&, std::tuple<>&&)':
FrequencyDialog.cpp:(.text._ZNSt8_Rb_treeINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4pairIKS5_N8SoapySDR5RangeEESt10_Select1stISA_ESt4lessIS5_ESaISA_EE22_M_emplace_hint_uniqueIJRKSt21piecewise_construct_tSt5tupleIJRS7_EESL_IJEEEEESt17_Rb_tree_iteratorISA_ESt23_Rb_tree_const_iteratorISA_EDpOT_[_ZNSt8_Rb_treeINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4pairIKS5_N8SoapySDR5RangeEESt10_Select1stISA_ESt4lessIS5_ESaISA_EE22_M_emplace_hint_uniqueIJRKSt21piecewise_construct_tSt5tupleIJRS7_EESL_IJEEEEESt17_Rb_tree_iteratorISA_ESt23_Rb_tree_const_iteratorISA_EDpOT_]+0x8f): undefined reference to `SoapySDR::Range::Range()'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::setSoapyDevice(SoapySDR::Device*)':
SDRDeviceInfo.cpp:(.text+0x799): undefined reference to `SoapySDR::Device::unmake(SoapySDR::Device*)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getSoapyDevice()':
SDRDeviceInfo.cpp:(.text+0x7cc): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::hasCORR(int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0x9f0): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getAntennaNames[abi:cxx11](int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0xa71): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getAntennaName[abi:cxx11](int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0xb16): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getCurrentGain(int, unsigned long, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)':
SDRDeviceInfo.cpp:(.text+0xd9b): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::~SDRDeviceInfo()':
SDRDeviceInfo.cpp:(.text+0xf2a): undefined reference to `SoapySDR::Device::unmake(SoapySDR::Device*)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getSampleRates(int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0x17e6): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o: in function `SDRDeviceInfo::getGains[abi:cxx11](int, unsigned long)':
SDRDeviceInfo.cpp:(.text+0x1e95): undefined reference to `SoapySDR::Device::make(std::map<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::less<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<std::pair<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > > const&)'
/usr/bin/ld: CMakeFiles/CubicSDR.dir/src/sdr/SDREnumerator.cpp.o: in function `SDREnumerator::enumerate_devices(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, bool)':
SDREnumerator.cpp:(.text+0x2a4b): undefined reference to `SoapySDR::getAPIVersion[abi:cxx11]()'
/usr/bin/ld: SDREnumerator.cpp:(.text+0x2ade): undefined reference to `SoapySDR::getABIVersion[abi:cxx11]()'
/usr/bin/ld: SDREnumerator.cpp:(.text+0x2b71): undefined reference to `SoapySDR::getRootPath[abi:cxx11]()'
/usr/bin/ld: SDREnumerator.cpp:(.text+0x2cd4): undefined reference to `SoapySDR::listModules(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: SDREnumerator.cpp:(.text+0x2e9a): undefined reference to `SoapySDR::loadModule(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
/usr/bin/ld: SDREnumerator.cpp:(.text+0x2f53): undefined reference to `SoapySDR::Registry::listFindFunctions[abi:cxx11]()'
[... cut ...] 5_N8SoapySDR7ArgInfoEESt10_Select1stISA_ESt4lessIS5_ESaISA_EE22_M_emplace_hint_uniqueIJRKSt21piecewise_construct_tSt5tupleIJRS7_EESL_IJEEEEESt17_Rb_tree_iteratorISA_ESt23_Rb_tree_const_iteratorISA_EDpOT_]+0x4e): undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1398: x64/CubicSDR] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

SDRplay, 2018-12-05:
undefined reference to SoapySDR - do you have SoapySDR built and installed (complete with sudo ldconfig)?

krumelmonster, 2018-12-05:
I seem to have the same issue using the [cubicsdr-git PKGBUILD from arch aur](https://aur.archlinux.org/packages/cubicsdr-git/) on an up-to-date arch system.
The output of make VERBOSE=1 is:

	/usr/bin/cmake -H/home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git -B/home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build --check-build-system CMakeFiles/Makefile.cmake 0
	/usr/bin/cmake -E cmake_progress_start /home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build/CMakeFiles /home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build/CMakeFiles/progress.marks
	make -f CMakeFiles/Makefile2 all
	make[1]: Entering directory '/home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build'
	make -f CMakeFiles/CubicSDR.dir/build.make CMakeFiles/CubicSDR.dir/depend
	make[2]: Entering directory '/home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build'
	cd /home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git /home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git /home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build /home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build /home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build/CMakeFiles/CubicSDR.dir/DependInfo.cmake --color=
	make[2]: Leaving directory '/home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build'
	make -f CMakeFiles/CubicSDR.dir/build.make CMakeFiles/CubicSDR.dir/build
	make[2]: Entering directory '/home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build'
	[  1%] Linking CXX executable x64/CubicSDR
	/usr/bin/cmake -E cmake_link_script CMakeFiles/CubicSDR.dir/link.txt --verbose=1
	/usr/lib/ccache/bin/c++  -march=native -O2 -pipe -fstack-protector-strong -fno-plt -pthread -O3 -DNDEBUG  -Wl,-O1,--sort-common,--as-needed,-z,relro,-z,now -rdynamic CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o CMakeFiles/CubicSDR.dir/src/AppConfig.cpp.o CMakeFiles/CubicSDR.dir/src/FrequencyDialog.cpp.o CMakeFiles/CubicSDR.dir/src/DemodLabelDialog.cpp.o CMakeFiles/CubicSDR.dir/src/IOThread.cpp.o CMakeFiles/CubicSDR.dir/src/ModemProperties.cpp.o CMakeFiles/CubicSDR.dir/src/BookmarkMgr.cpp.o CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o CMakeFiles/CubicSDR.dir/src/sdr/SDRPostThread.cpp.o CMakeFiles/CubicSDR.dir/src/sdr/SDREnumerator.cpp.o CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o CMakeFiles/CubicSDR.dir/src/demod/DemodulatorThread.cpp.o CMakeFiles/CubicSDR.dir/src/demod/DemodulatorWorkerThread.cpp.o CMakeFiles/CubicSDR.dir/src/demod/DemodulatorInstance.cpp.o CMakeFiles/CubicSDR.dir/src/demod/DemodulatorMgr.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/Modem.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/ModemAnalog.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/ModemDigital.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemDSB.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemFM.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemNBFM.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemFMStereo.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemIQ.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemLSB.cpp.o CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemUSB.cpp.o CMakeFiles/CubicSDR.dir/src/audio/AudioThread.cpp.o CMakeFiles/CubicSDR.dir/src/audio/AudioSinkThread.cpp.o CMakeFiles/CubicSDR.dir/src/audio/AudioSinkFileThread.cpp.o CMakeFiles/CubicSDR.dir/src/audio/AudioFile.cpp.o CMakeFiles/CubicSDR.dir/src/audio/AudioFileWAV.cpp.o CMakeFiles/CubicSDR.dir/src/util/Gradient.cpp.o CMakeFiles/CubicSDR.dir/src/util/Timer.cpp.o CMakeFiles/CubicSDR.dir/src/util/MouseTracker.cpp.o CMakeFiles/CubicSDR.dir/src/util/GLExt.cpp.o CMakeFiles/CubicSDR.dir/src/util/GLFont.cpp.o CMakeFiles/CubicSDR.dir/src/util/DataTree.cpp.o CMakeFiles/CubicSDR.dir/src/panel/ScopePanel.cpp.o CMakeFiles/CubicSDR.dir/src/panel/SpectrumPanel.cpp.o CMakeFiles/CubicSDR.dir/src/panel/WaterfallPanel.cpp.o CMakeFiles/CubicSDR.dir/src/panel/MeterPanel.cpp.o CMakeFiles/CubicSDR.dir/src/visual/ColorTheme.cpp.o CMakeFiles/CubicSDR.dir/src/visual/PrimaryGLContext.cpp.o CMakeFiles/CubicSDR.dir/src/visual/InteractiveCanvas.cpp.o CMakeFiles/CubicSDR.dir/src/visual/MeterCanvas.cpp.o CMakeFiles/CubicSDR.dir/src/visual/MeterContext.cpp.o CMakeFiles/CubicSDR.dir/src/visual/TuningCanvas.cpp.o CMakeFiles/CubicSDR.dir/src/visual/TuningContext.cpp.o CMakeFiles/CubicSDR.dir/src/visual/ModeSelectorCanvas.cpp.o CMakeFiles/CubicSDR.dir/src/visual/ModeSelectorContext.cpp.o CMakeFiles/CubicSDR.dir/src/visual/ScopeCanvas.cpp.o CMakeFiles/CubicSDR.dir/src/visual/ScopeContext.cpp.o CMakeFiles/CubicSDR.dir/src/visual/SpectrumCanvas.cpp.o CMakeFiles/CubicSDR.dir/src/visual/WaterfallCanvas.cpp.o CMakeFiles/CubicSDR.dir/src/visual/GainCanvas.cpp.o CMakeFiles/CubicSDR.dir/src/visual/ImagePanel.cpp.o CMakeFiles/CubicSDR.dir/src/process/VisualProcessor.cpp.o CMakeFiles/CubicSDR.dir/src/process/ScopeVisualProcessor.cpp.o CMakeFiles/CubicSDR.dir/src/process/SpectrumVisualProcessor.cpp.o CMakeFiles/CubicSDR.dir/src/process/FFTVisualDataThread.cpp.o CMakeFiles/CubicSDR.dir/src/process/FFTDataDistributor.cpp.o CMakeFiles/CubicSDR.dir/src/process/SpectrumVisualDataThread.cpp.o CMakeFiles/CubicSDR.dir/src/ui/GLPanel.cpp.o CMakeFiles/CubicSDR.dir/src/forms/SDRDevices/SDRDevices.cpp.o  [... cut ...] st_iteratorISA_EDpOT_]+0x50): undefined reference to `SoapySDR::ArgInfo::ArgInfo()'
	collect2: error: ld returned 1 exit status
	make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:1393: x64/CubicSDR] Error 1
	make[2]: Leaving directory '/home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build'
	make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/CubicSDR.dir/all] Error 2
	make[1]: Leaving directory '/home/jo/.cache/pacaur/cubicsdr-git/src/cubicsdr-git/build'
	make: *** [Makefile:130: all] Error 2