cjcliffe/CubicSDR #694: Memory Database System Is Essentially Unusable

Robert1111x, 2018-12-08:
Hello,

This probably goes under the "wish list" and hoped for category.
Wonderful program.
Thanks so much for making it available.
Must have taken an incredible amount of work; great interface, etc.

But the "Memory" system implementation for saving favorite frequencies, perhaps
downloading and using eibi, etc. is truly poor.  
Essentially overly complicated, convoluted, and unusable.

A more typical and easy to read "database" filing system, and means of saving and recalling favorite frequencies (and modes) would be a tremendous help. Even something simple, but straightforward and more typical of what other similar programs have.

I'm using on a Mac.

Anything along this line "presently in the works" ?

Thanks,
Bob

cjcliffe, 2019-01-03:
@Robert1111x ahoy, just want to clarify what version are you using?  

There's been a bookmarks feature in CubicSDR for some time now; is that what you're referring to as the 'memory' implementation?

Robert1111x, 2019-01-03:
Hello,

Thanks for help; much appreciated.

Using the latest version, I think, which shows as "SDR 2"
Running it on an iMac under Mojave.

Please don't take any of this as criticisms; I love the program.
Wonderful program; thanks for making it available.

Am a Sr. Citizen now, very, and I have to admit that I miss a lot of the 
nuances in programs like this, I'm afraid.

a.  Yes, it is the Bookmarks, I was referring too.

For some future revision, might it not be possible to have
a straightforward, simple, memory database structure like
so many of the other radio programs have incorporated ?
Look at SDRUno, and many others, e.g.

I have a devil of a time, frankly, in using Bookmark to save and go to later.
Find it, frankly, very "convoluted," hard to use, and non-intuitive (that old age thing, probably).

Would also be great to be able to import, easily, eibi into.

b.  Might I also ask please:  I use with SDRPlay.
But most of the time it doesn't appear in SDR Devices when I open it up.
Most of the time I have to try a lot of 'things" to get it to appear.
Under SDR Devices/Normal I often get "Not Found (SDR Play)"
Selecting SDR Play (in the right hand Soapy column) doesn't help, usually.

What's the "secret" to have it always appear and automatically
open up ?  As usual, I'm missing something.
Would be grateful for any hints on how to go about it, so it always works.

Again, a terrific program.

Thanks again for thoughts and help,
Bob  (outside of Boston)
----------------

cjcliffe, 2019-01-04:
@Robert1111x Thanks for your suggestions

a.)  I agree the bookmarks aren't very intuitive; I built the simplest version I could and hope to improve it in the future.  

I haven't improved it yet since I believe I'm going to change the interface that CubicSDR uses (currently wxWidgets) to something which will provide a better experience.

I'll try to remember to update this issue when there's something new for you to try.

b.)  I'm not sure I've experienced that issue with my own SDRPlay devices, but if you're able to take a screenshot picture of the 'SDR Devices' window showing the 'Not Found (SDR Play)' message I might be able to help diagnose it further.   

Right now it just sounds like you've added SDRPlay manually to the devices list and it's only showing as connected sometimes, but I want to be sure of what you're seeing with a screenshot.

From my experience if I have the SDRPlay cable plugged into a USB hub device instead of directly into the computer's USB port it sometimes has problems finding it.   So if you've got a hub between the SDRPlay and the computer, then maybe try plugging SDRPlay cable directly to the computer instead to see if that helps you.

Cheers

Robert1111x, 2019-01-04:
Hi Charles,

Thanks so much for offer of help.

Seems like my kid removed my USB connection to the SDRPlay so he could
use the Port for one of his games.
Never noticed it !

All seems fine now.
Again, thanks so much for the help.

Wonderful program, and I can just imagine how much work went into it.

Sure is nice being able to run it directly in the iMac OS without having
to bother with Parallels or Bootcamp.

Please let me know if and when you have a new version available,
with or without a revised "memory" scheme as I suggested.
Even if just a simple way to import eibi would be great.

Am in my 80's now, and frankly I don't look at GibHub very often.
Do keep me in mind. Would appreciate your letting me directly know of any new rev's.

Best regards, and wishes for the New Year,
Bob
------------------