cjcliffe/CubicSDR #695: Often not able to detect short bursts of RF.

gpgib6, 2018-12-19:
In this picture, the waterfall shows the same relatively strong signal but only in the right picture was the energy properly picked up and displayed in the frequency domain. I would love to know how to make this more reliable if possible. Not sure if it is because of hackrf one or cubesdr.
[image]

cjcliffe, 2019-01-03:
@gpgib6 you can increase the speed of the waterfall with the slider to the right but there's currently no averaging -- it simply shows a sample from the time the waterfall stepped.

I'll add this as a feature request since it should technically be possible to add accumulation support to the waterfall processor.

gpgib6, 2019-01-13:
Thanks, that would be pretty awesome.

gpgib6, 2020-09-02:
Just a friendly reminder that this bug is still happening. In the screenshot I can see RF energy in the waterfall display but the peak trace in the frequency domain missed it. It would be nice to be able to walk away from the test and rely on CubicSDR to catch any RF energy that gets picked up by the hackRF one.

[image]