cjcliffe/CubicSDR #696: CubicSDR + LinuxMint 19.1 + SDRplay RSP1A

opabert, 2018-12-25:
Dear team,
Contact SDRplay support they pointed me to this forum.
Like asking i fallow this video instruction
http://kb9rlw.blogspot.com/2017/06/step-by-step-install-of-sdrplay-rsp2.html
After install try start CubicSDR in terminal type CubicSDR get this message

The program 'CubicSDR' received an X Window System error.
This probably reflects a bug in the program.
The error was 'BadValue (integer parameter out of range for operation)'.
  (Details: serial 217 error_code 2 request_code 155 minor_code 3)
  (Note to programmers: normally, X errors are reported asynchronously;
   that is, you will receive the error a while after causing it.
   To debug your program, run it with the --sync command line
   option to change this behavior. You can then get a meaningful
   backtrace from your debugger if you break on the gdk_x_error() function.)

vsonnier, 2019-01-23:
@opabert Hi ! Looks like a problem we saw before, likely related to a buggy wxWidgets. 
You should try using the latest wxWidgets 3.12.
Since you used https://github.com/loughkb/SDRplaylinux scripts, here are mines:

Install_script-pt1.sh:
```bash
#! /bin/bash
# This script should install and build all needed stuff for
# using the SDRPlay on Ubuntu 16.04 or later.  It must be
# run as root.  
# 
# First we install needed dependancies.

#apt-get install git build-essential automake cmake g++ swig
#apt-get install libgtk-3-dev libpulse-dev libpython-dev python-numpy
#apt-get install mesa-utils libeglw1-mesa libglw1-mesa-dev
#apt-get install freeglut3-dev freeglut3

apt-get install git build-essential automake cmake g++  mesa-utils libgtk-3-dev freeglut3-dev freeglut3

# Now we'll git the projects we need

git clone https://github.com/jgaeddert/liquid-dsp
wget https://github.com/wxWidgets/wxWidgets/releases/download/v3.1.2/wxWidgets-3.1.2.tar.bz2
tar -xvjf wxWidgets-3.1.2.tar.bz2  
git clone https://github.com/pothosware/SoapySDR.git
git clone https://github.com/pothosware/SoapySDRPlay.git
git clone https://github.com/pothosware/SoapyRemote.git
git clone https://github.com/cjcliffe/CubicSDR.git

# Build liquid-dsp

cd liquid-dsp
./bootstrap.sh
./configure --enable-fftoverride 
make -j4
make install
ldconfig
cd ..

# Build wxwidgets

cd wxWidgets-3.1.2/
mkdir -p ~/Develop/wxWidgets-staticlib
./autogen.sh 
./configure --with-opengl --disable-shared --enable-monolithic --with-libjpeg --with-libtiff --with-libpng --with-zlib --disable-sdltest --enable-unicode --enable-display --enable-propgrid --disable-webkit --disable-webview --disable-webviewwebkit --prefix=`echo ~/Develop/wxWidgets-staticlib` CXXFLAGS="-std=c++0x" --with-libiconv=/usr
make -j8 
make install
cd ..

echo "."
echo "First part of installation done.   Now run the SDRPlay file you"
echo "downloaded from their site.  Be sure to use sudo to run it!"
```

Install_script-pt12.sh:
```bash
#! /bin/bash
# This script should install and build all needed stuff for
# using the SDRPlay on Ubuntu 16.04 or later.  It must be
# run as root.  
# 
# This is part two.
# Here we build SoapySDR

cd SoapySDR
mkdir build
cd build
cmake ..
make -j8
make install
ldconfig
cd ..
cd ..

# Now we build the SDRPlay module for Soapy

cd SoapySDRPlay
mkdir build
cd build
cmake ..
make
make install
cd ..
cd ..

# And we build SoapyRemote

cd SoapyRemote
mkdir build
cd build
cmake ..
make
make install
cd ..
cd ..

# And finally,  we build Cubic.  This takes awhile!

cd CubicSDR
mkdir build
cd build
cmake ../ -DCMAKE_BUILD_TYPE=Release -DwxWidgets_CONFIG_EXECUTABLE=~/Develop/wxWidgets-staticlib/bin/wx-config
make
make install
cd ..
cd ..
rm -R ~/Develop

# now we change permissions on these root-owned folders so the user can 
# delete them at their leisure.

chmod -R 0777 ./*

# And we're done.   Cubic should work with the SDRPlay.  The user can
# run it from the terminal with CubicSDR or add this command to a menu.
```

Works for me on Ubuntu 18.04 running under WSL (Windows Subsystem For Linux) the Windows 10 Linux kernel emulation.

ClaudeMa, 2019-04-17:
Version 0.2.5 build from sources work for me on Mint 19.1 without any other issues that one that is already know

cjcliffe, 2019-05-24:
Verified original scripts referenced are using wxWidgets-3.1.0 and just need to be updated to 3.1.2

closed 2019-05-24 by cjcliffe