cjcliffe/CubicSDR #697: AppImage fails on linux with symbol lookup error

solsticedhiver, 2019-01-04:
On ubuntu 18.10, trying to run the appimage gives the following error:

```
[ERROR] SoapySDR::loadModule(.//////////lib/SoapySDR/modules0.7/libsdrPlaySupport.so)
  dlopen() failed: libmirsdrapi-rsp.so.2.13: cannot open shared object file: No such file or directory
	Available factories...airspy, airspyhf, audio, bladerf, hackrf, lime, null, redpitaya, remote, rtlsdr
./CubicSDR-0.2.5-x86_64.AppImage: symbol lookup error: /usr/lib/x86_64-linux-gnu/libGLX_mesa.so.0: undefined symbol: xcb_dri3_get_supported_modifiers
```
One gets briefly the GUI appearing then closing.

That's 2 errors:
- one for the non existent libmirsdrapi-rsp.so lib.
- one for the symbol lookup error in libGLX_msea.so.

cjcliffe, 2019-01-08:
@solsticedhiver this may be a known issue from what I've quickly googled.

Does it work if you run it like this?

`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libxcb-dri3.so.0 ./CubicSDR-0.2.5-x86_64.AppImage`

solsticedhiver, 2019-01-08:
@cjcliffe yes that works. thanks

closed 2019-05-24 by cjcliffe

mjpowersjr, 2020-01-01:
I can confirm this works for me using Ubuntu 19.10 as well.

RepComm, 2020-11-20:
Can confirm fix working using Ubuntu 20.04LTS

Abrahammar997, 2021-09-28:
Don't work on Ubuntu 21.04. It runs but when "snips" it make window black