cjcliffe/CubicSDR #702: CubicSDR issues with Pulseaudio output

Tnix76, 2019-01-16:
Hi all,
I have CubicSDR 0.2.5 supplied by my Linux distro Debian Buster. It runs fine and detects my SDRPlay RSP2 but I'm afraid that CubicSDR can only sees ALSA devices but cannot work with Pulseaudio. When I select a modem I can only see the next options:

hw:HDA Intel HDMI,7
hw:HDA Intel HDMI,8
hw:HDA Intel HDMI,9
hw:HDA Intel HDMI,10
hw:HDA Intel PCH,0

[image]

In my case I haven't any HDMI device connected and I work with the last one.

With this I cannot pipe the audio with Pulseaudio to decode for example.

Next I paste the output when I startup CubicSDR in console and select my SDRPlay device:

`Loading:: configuration file '/home/user/.CubicSDR/config.xml'
Loaded 230 rig models via hamlib.

Audio Device #0 hw:HDA Intel HDMI,3
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 8
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                32-bit signed integer.
        Supported sample rates:
                32000hz
                44100hz
                48000hz
                88200hz
                96000hz
                176400hz
                192000hz

Audio Device #1 hw:HDA Intel HDMI,7
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 8
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                32-bit signed integer.
        Supported sample rates:
                32000hz
                44100hz
                48000hz
                88200hz
                96000hz
                176400hz
                192000hz

Audio Device #2 hw:HDA Intel HDMI,8
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 8
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                32-bit signed integer.
        Supported sample rates:
                32000hz
                44100hz
                48000hz
                88200hz
                96000hz
                176400hz
                192000hz

Audio Device #3 hw:HDA Intel HDMI,9
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 8
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                32-bit signed integer.
        Supported sample rates:
                32000hz
                44100hz
                48000hz
                88200hz
                96000hz
                176400hz
                192000hz

Audio Device #4 hw:HDA Intel HDMI,10
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 8
        Duplex channels: 0
        Native formats:
                16-bit signed integer.
                32-bit signed integer.
        Supported sample rates:
                32000hz
                44100hz
                48000hz
                88200hz
                96000hz
                176400hz
                192000hz

Audio Device #5 hw:HDA Intel PCH,0
        Default Output? No
        Default Input? No
        Input channels: 2
        Output channels: 2
        Duplex channels: 2
        Native formats:
                16-bit signed integer.
                32-bit signed integer.
        Supported sample rates:
                44100hz
                48000hz
                96000hz
                192000hz

ALSA lib pcm_dmix.c:1099:(snd_pcm_dmix_open) unable to open slave

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No existe el fichero o el directorio.

ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No existe el fichero o el directorio.

Audio Device #6 
        Default Output? No
        Default Input? No
        Input channels: 0
        Output channels: 0
        Duplex channels: 0
        Native formats:
        Supported sample rates:

Available vertical sync SwapInterval functions: 
        glxSwapIntervalEXT: Yes
        DRI2SwapInterval: No
        glxSwapIntervalMESA: No
        glxSwapIntervalSGI: Yes
Using glxSwapIntervalEXT.

Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono12_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono16_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono18_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono24_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono27_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono32_0.png', parsed 255 characters.
Loaded font 'Bitstream Vera Sans Mono' from '/usr/share/cubicsdr/fonts/vera_sans_mono36_0.png', parsed 255 characters.
SDR enumerator starting.
SoapySDR init..
        API Version: v0.6.0
        ABI Version: v0.6
        Install root: /usr
        Loading modules... 
        Available factories...airspy, audio, bladerf, hackrf, lime, miri, null, osmosdr, redpitaya, remote, rfspace, rtlsdr, sdrplay, uhd
ALSA lib pcm_dmix.c:1099:(snd_pcm_dmix_open) unable to open slave

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No existe el fichero o el directorio.

ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No existe el fichero o el directorio.

[INFO] [UHD] linux; GNU C++ version 8.2.0; Boost_106700; UHD_3.13.0.2-4+b1
  default_input = False
  default_output = False
  device_id = 5
  driver = audio
  label = hw:HDA Intel PCH,0
Make device 0
ALSA lib pcm_dmix.c:1099:(snd_pcm_dmix_open) unable to open slave

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No existe el fichero o el directorio.

ALSA lib pcm_dsnoop.c:638:(snd_pcm_dsnoop_open) unable to open slave

RtApiAlsa::getDeviceInfo: snd_pcm_open error for device (default), No existe el fichero o el directorio.

  device_id=5
  origin=https://github.com/pothosware/SoapyAudio

  driver = sdrplay
  label = SDRplay Dev0 RSP2 1710015F20
Make device 1
  mir_sdr_api_version=2,130000
  mir_sdr_hw_version=2

Reporting enumeration complete.
SDR enumerator done.
SDR thread starting.
device init()
[INFO] Using format CF32.
Device Stream MTU: 65536
starting readLoop()
`
Thanks for your help.
Regards.

Tnix76, 2019-01-16:
Some error messages translated:

"No existe el fichero o el directorio" = "No such file or directory"

Tnix76, 2019-01-27:
Yes, other Pulseaudio applications work fine. I need to redirect CubicSDR sound to Dream DRM decoder but I'm afraid Dream only works with Pulseaudio devices so I'm trying to recompile Dream with ALSA support.

vsonnier, 2019-01-28:
Hi @Tnix76, this is probably related to this problem: https://github.com/thestk/rtaudio/pull/129.
This may be related to #620.
So, try to re-compile CubicSDR with PulseAudio-only backend for RtAudio:

https://github.com/cjcliffe/CubicSDR/issues/620#issuecomment-368291020

cjcliffe, 2019-05-24:
@Tnix76 RTAudio has been updated by @vsonnier recently and the fix will be included in 0.2.6 releases; thanks!

closed 2019-05-24 by cjcliffe