cjcliffe/CubicSDR #703: cubicsdr0.2.5 on Debian Stretch - Unable to enter serial port

ajcher, 2019-01-18:
Built CubicSDR on Debian Stretch and got it working. Next step is to get cubicsdr to talk with rigctld so cubicsdr can follow the rig. Unable to change (i.e. manually enter) serial port info (refer image). 

In CubicSDR, RigControl->Enable Rig is selected and the RigControl->Model is set to Hamlib Net Rig Ctl. Need to enter 127.0.0.1:nnnn. Is there a way to do it in the config file?

[image]

Thx a lot.

cjcliffe, 2019-01-21:
@ajcher there should be a text box just below the list where you can type manually; does it appear if you resize the window?

cjcliffe, 2019-01-23:
@ajcher ah, yes that should be resizable; and probably larger by default.  Thanks for the follow-up.

ajcher, 2019-01-24:
Found a workaround ... by  forcing the ip:port sequence in the config file...but the config file gets over-written each time i use the pgm. The way out was to make the necessary changes in settings, quit pgm, hard code the ip:port sequence and change the file permission to RO. Which means, it kinda-sorta works and am functional for now. Thx for all the effort you put in to keep this going. Much appreciated ...

cjcliffe, 2019-05-09:
@ajcher latest merge to master branch now includes a fixed port selection dialog with resize supported; you should now be able to use the text box below the entry list to set the ip:port

closed 2019-05-24 by cjcliffe