cjcliffe/CubicSDR #705: AppImage saving setting to incorrect? location

warren-axiell, 2019-01-20:
Hello, 

I am using your .AppImage version which is fantastic, thanks!  However, it was not saving any of my settings.  I ran strace on it and found it was trying to read them from ~/.CubicSDR-0.2.4-x86_64/ but was writing them to ~/.CubicSDR/

I ended up symlinking the two and everything now works fine. 

Thanks!

cjcliffe, 2019-01-21:
@warren-axiell interesting, it should only be using ~/.CubicSDR/ -- thanks for the report.

cjcliffe, 2019-05-14:
@warren-axiell I think I have it fixed now; here's the AppImage test build if you'd like to confirm:

[CubicSDR-0.2.5-x86_64.zip](https://github.com/cjcliffe/CubicSDR/files/3175624/CubicSDR-0.2.5-x86_64.zip)

closed 2019-05-24 by cjcliffe