cjcliffe/CubicSDR #706: Crashes when config files are present

yurivict, 2019-02-09:
It runs fine the first time, but crashes the second time when it parses configs that the first run wrote:

```
$ CubicSDR 
Loading:: configuration file '/home/yuri/.CubicSDR/config.xml'
LoadFromFileXML[error loading]: /home/yuri/.CubicSDR/config.xml
Segmentation fault
```

FreeBSD 11.2 amd64

vsonnier, 2019-02-10:
@yurivict Hiello ! A backtrace on  a debug build would be greatly appreciated because obviously It Works For Me :)

yurivict, 2019-02-10:
```
Starting program: /usr/local/bin/CubicSDR 
Loading:: configuration file '/home/yuri/.CubicSDR/config.xml'
LoadFromFileXML[error loading]: /home/yuri/.CubicSDR/config.xml

Program received signal SIGSEGV, Segmentation fault.
__je_arena_mapbits_get (chunk=0x0, pageind=<optimized out>) at /usr/src/contrib/jemalloc/include/jemalloc/internal/arena.h:806
806		return (arena_mapbitsp_read(arena_mapbitsp_get_const(chunk, pageind)));
(gdb) bt
#0  0x0000000803f1e939 in __je_arena_mapbits_get (chunk=0x0, pageind=<optimized out>) at /usr/src/contrib/jemalloc/include/jemalloc/internal/arena.h:806
#1  0x0000000803f1e939 in __je_arena_mapbits_binind_get (chunk=0x0, pageind=<optimized out>) at /usr/src/contrib/jemalloc/include/jemalloc/internal/arena.h:863
#2  0x0000000803f1e939 in __je_arena_salloc (demote=false, tsdn=<optimized out>, ptr=<optimized out>) at /usr/src/contrib/jemalloc/include/jemalloc/internal/arena.h:1384
#3  0x0000000803f1e939 in __je_isalloc (demote=false, tsdn=<optimized out>, ptr=<optimized out>)
    at /usr/src/contrib/jemalloc/include/jemalloc/internal/jemalloc_internal.h:951
#4  0x0000000803f1e939 in ifree (tsd=0x0, ptr=<optimized out>, tcache=<optimized out>, slow_path=<optimized out>) at jemalloc_jemalloc.c:1810
#5  0x0000000803f1efd1 in __free (ptr=0x1) at jemalloc_jemalloc.c:1935
#6  0x0000000000644664 in TiXmlDocument::~TiXmlDocument() (this=0x7fffffffbe78) at /usr/local/include/tinyxml.h:1411
#7  0x000000000063dc61 in DataTree::LoadFromFileXML(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, DT_FloatingPointPolicy)
    (this=0x7fffffffcab8, filename=..., fpp=USE_FLOAT) at /usr/ports/comms/cubicsdr/work/CubicSDR-0.2.5-38-gb0505b0/src/util/DataTree.cpp:1708
#8  0x00000000004a8e2e in AppConfig::load() (this=0x811e9da98) at /usr/ports/comms/cubicsdr/work/CubicSDR-0.2.5-38-gb0505b0/src/AppConfig.cpp:684
#9  0x000000000043b7dc in CubicSDR::OnCmdLineParsed(wxCmdLineParser&) (this=0x811e9d800, parser=...)
    at /usr/ports/comms/cubicsdr/work/CubicSDR-0.2.5-38-gb0505b0/src/CubicSDR.cpp:531
#10 0x000000080206e297 in wxAppConsoleBase::OnInit() () at /usr/local/lib/libwx_baseu-3.1.so.2
#11 0x0000000000437e7a in CubicSDR::OnInit() (this=0x811e9d800) at /usr/ports/comms/cubicsdr/work/CubicSDR-0.2.5-38-gb0505b0/src/CubicSDR.cpp:238
#12 0x0000000000446b69 in wxAppConsoleBase::CallOnInit() (this=0x811e9d800) at /usr/local/include/wx-3.1/wx/app.h:93
#13 0x0000000802105f93 in wxEntry(int&, wchar_t**) () at /usr/local/lib/libwx_baseu-3.1.so.2
#14 0x0000000000432fe6 in main(int, char**) (argc=1, argv=0x7fffffffe800) at /usr/ports/comms/cubicsdr/work/CubicSDR-0.2.5-38-gb0505b0/src/CubicSDR.cpp:28
```

vsonnier, 2019-02-11:
Thanks. Please add the compiling command line, and the ``config.xml` that created the crash.
According to your backtrace, following some error in the `config.xml` the virtual destructor `TiXmlDocument::~TiXmlDocument()` is called and creates the crash, because its pointer `this` is invalid.

From this, I cannot say more and don't see why  the `TiXmlDocument` would get corrupted. Maybe having your `config.xml` would help.

yurivict, 2019-02-11:
```config.xml``` is empty:

```
ls -l /home/yuri/.CubicSDR/
total 0
-rw-r--r--  1 yuri  users  0 Feb  9 10:00 config.xml
```

vsonnier, 2019-02-11:
Thanks again 👍 So the root problem was that the `DataTree::SaveToFileXML()` failed because at some point TinyXML used a buffer overflow `strlen`. While attempting a `<![CDATA[%s]]>` which we don't use in CubicSDR anyway.

Hem. I'll try to investigate later I hope the bug is not in TinyXML itself 👎

vsonnier, 2019-02-20:
I've seen no trace in our code where the `TiXmlText::cdata` could be set to `true`, so what I'm supposing is that the field go smashed by our `DataTree` wrapper around `TinyXML`  using one `memcpy` too many.

Please test this branch: https://github.com/cjcliffe/CubicSDR/tree/vso_dataelement_rework
where I've rewritten `DataElement` handling using templates and `vector` of bytes instead of raw pointer handling. Tell me if it's better for you.

I've also removed serialization support from `DataTree` to lighten the maintainance burden.

vsonnier, 2019-02-21:
Forced-push another version into https://github.com/cjcliffe/CubicSDR/tree/vso_dataelement_rework, fix a crash on empty string XML tag (by re-introducing the old code), also removed `long double` support which is not a distinct type on all machines and which support was incomplete anyway.

vsonnier, 2019-02-23:
I __think__ I've got the thing right this time, could you please  @yurivict test the branch https://github.com/cjcliffe/CubicSDR/tree/vso_dataelement_rework ? Thanks !

Mur77, 2019-02-28:
Hello,

I'm unable to build this branch, because of errors:
`[  1%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o`
`In file included from /home/mur/pf/CubicSDR-vso/src/CubicSDR.cpp:17:`
`In file included from /home/mur/pf/CubicSDR-vso/src/CubicSDR.h:16:`
`In file included from /home/mur/pf/CubicSDR-vso/src/sdr/SoapySDRThread.h:11:`
`In file included from /home/mur/pf/CubicSDR-vso/src/AppConfig.h:13:`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:134:25: error: explicit`
`      specialization of 'determineScalarDataType' in class scope`
`    DataElementTypeEnum determineScalarDataType(const char& type_in) { r...`
`____________________^`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:137:25: error: explicit`
`      specialization of 'determineScalarDataType' in class scope`
`    DataElementTypeEnum determineScalarDataType(const unsigned char& typ...`
`____________________^`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:140:25: error: explicit`
`      specialization of 'determineScalarDataType' in class scope`
`    DataElementTypeEnum determineScalarDataType(const int& type_in) { re...`
`____________________^`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:143:25: error: explicit`
`      specialization of 'determineScalarDataType' in class scope`
`    DataElementTypeEnum determineScalarDataType(const unsigned int& type...`

There are some more of such errors!

vsonnier, 2019-02-28:
@Mur77  Aaaah Templates ! I push a commit on the branch to try to fix this, tell me if it works.

Mur77, 2019-03-01:
Hi!

I got similar errors at the other place:
`[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o`
`In file included from /home/mur/pf/CubicSDR-vso/src/CubicSDR.cpp:17:`
`In file included from /home/mur/pf/CubicSDR-vso/src/CubicSDR.h:16:`
`In file included from /home/mur/pf/CubicSDR-vso/src/sdr/SoapySDRThread.h:11:`
`In file included from /home/mur/pf/CubicSDR-vso/src/AppConfig.h:13:`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:248:10: error: explicit`
`      specialization of 'set' in class scope`
`    void set(const string& str_in) {`
`_____^`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:257:10: error: explicit`
`      specialization of 'set' in class scope`
`    void set(const wstring& wstr_in) {`
`_____^`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:278:10: error: explicit`
`      specialization of 'set' in class scope`
`    void set(const vector<string>& vector_str_in) {`
`_____^`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:417:10: error: explicit`
`      specialization of 'get' in class scope`
`    void get(string& str_out) {`
`_____^`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:439:10: error: explicit`
`      specialization of 'get' in class scope`
`    void get(wstring& wstr_out) {`
`_____^`
`/home/mur/pf/CubicSDR-vso/src/util/DataTree.h:474:10: error: explicit`
`      specialization of 'get' in class scope`
`    void get(vector<string>& vector_str_out) {`
`_____^`
`6 errors generated.`

vsonnier, 2019-03-01:
I forgot some fake partial templates last time. How about now ?

Mur77, 2019-03-02:
Hello, now I can build it and run successfully!

vsonnier, 2019-03-02:
Great news! So contrary to @yurivict, no trouble runing, saving settings, and so on ?

Mur77, 2019-03-02:
It runs without problems. Actually, I have one problem. It runs great. But when I click to waterfall to set demodulator frequency, the program slows down twice, like it's not enough CPU power, and sound is intermittent.
I'm not sure where is the problem. My CPU is Intel Celeron D 3.06 GHz.

vsonnier, 2019-03-02:
According to this site : https://www.cpubenchmark.net/cpu_list.php
Your CPU and mine has the following perfomance by comparison:
- Intel Celeron D 347 @ 3.06GHz : 270
- Intel Core i7-7700HQ @ 2.80GHz: 8822

Although mine is quite recent and powerful, yours is  indeed __very__ weak and has only 1 core.
CubicSDR is heavily multithreaded, and would need such a multicore processor. 

You can try to lower the Sample rate to the minimum possible for your device, and see where it goes.

Mur77, 2019-03-02:
I got it. At minimum rate (250 KHz) it produces stable sound.
Thank you!

vsonnier, 2019-03-02:
Thanks @Mur77 for your report. Sorry for your so weak machine, I wish you'll got fun with Cubic anayway.
I think this issue can be closed now.

closed 2019-03-02 by vsonnier