cjcliffe/CubicSDR #707: Cannot compie SoapyRTLSDR or Soapy SDRPlay modules

Twilight-Logic, 2019-02-15:
I can't seem to compile either if the two above modules on my Linux MINT 19 system. I am following the instructions to the letter. I get the following output:

> $~/src/CubicSDR/SoapySDRPlay/build $ cmake .. -DCMAKE_BUILD_TYPE=Release
> -- The CXX compiler identification is GNU 7.3.0
> -- Check for working CXX compiler: /usr/bin/c++
> -- Check for working CXX compiler: /usr/bin/c++ -- works
> -- Detecting CXX compiler ABI info
> -- Detecting CXX compiler ABI info - done
> -- Detecting CXX compile features
> -- Detecting CXX compile features - done
> CMake Error at CMakeLists.txt:7 (find_package):
>   Could not find a package configuration file provided by "SoapySDR"
>   (requested version 0.4.0) with any of the following names:
> 
>     SoapySDRConfig.cmake
>     soapysdr-config.cmake
> 
>   Add the installation prefix of "SoapySDR" to CMAKE_PREFIX_PATH or set
>   "SoapySDR_DIR" to a directory containing one of the above files.  If
>   "SoapySDR" provides a separate development package or SDK, be sure it has
>   been installed.
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/home/johnc/src/CubicSDR/SoapySDRPlay/build/CMakeFiles/CMakeOutput.log".

Dantali0n, 2019-02-15:
Cmake can't find the `SoapySDRConfig.cmake` or `soapysdr-config.cmake` files which it needs to determine if SoapySDR is installed and where. 
Likely the installation of SoapySDR itself was never executed or at least did not execute properly.

Twilight-Logic, 2019-02-15:
There is nothing mentioned in the instructions on that page about installing something called SoapySDR so I didn't do that installation step. If it was not installed by the CubicSDR package as a dependency, then the install was probably not executed. Please could this information be added to the instructions?

I did a search on 'soapysdr' and can see lots of related packages in the Ubuntu repository. Do I need libsoapysdr-dev or just the relevant module such as soapysdr0.6-module-rtlsdr? If I install the latter, do I still need to compile the sources I have downloaded?

Dantali0n, 2019-02-15:
Hi, I am sorry but your statement is incorrect. The documentation page you refer to very clearly lists installing SoapySDR. It is the fifth bold heading from the top named  `Build and install SoapySDR`

All the best,
Dantali0n

Twilight-Logic, 2019-02-15:
Sorry, I missed that. That step appears to be part of the main section of instructions for installing the Cubic SDR package. Since I had already installed the cubicsdr package from the repository using apt install cubicsdr, which should install all dependencies, I naturally went straight to the steps for installing the modules.

As it turns out, perhaps even this step is not required. I installed soapysdr-module-rtlsdr and that seems to have satisfied cubicsdr and it no longer complains about the missing module. So for RTLSDR it all boils down to two simple steps:

apt install cubicsdr
apt install soapysdr-module-rtlsdr

I notice there is no pre-compiled module for sdrplay. Is it no longer supported? Will I have to go through the whole compilation process just for that module anyway?

SDRplay, 2019-02-15:
After building/installing SoapySDR you need to build and install SoapySDRPlay (after you've installed the API from our website) - https://github.com/pothosware/SoapySDRPlay/wiki, it's still referenced on https://github.com/pothosware/SoapySDR/wiki - have you read that?

Twilight-Logic, 2019-02-15:
Ok, so you are saying I should not use any of the packages in the Ubuntu repository but compile instead?

Dantali0n, 2019-02-15:
> I notice there is no pre-compiled module for sdrplay. Is it no longer supported? Will I have to go through the whole compilation process just for that module anyway?

Unlikely, installing soapysdr-module-rtlsdr will also have installed the SoapySDR dependency. I suspect that compiling SoapySDRPlay will go fine but you do need to download a driver from the vendor. 

Furthermore, mixing packages from apt and compiling others manually from source can introduce issues. This is the reason we list liquid-dsp, wxwidgets and soapysdr installation instructions in the linux wiki as required steps to properly install cubicsdr. As you have now experienced yourself just using `apt install cucicsdr` and then continuing with the rest of the instructions already caused issues.

> Ok, so you are saying I should not use any of the packages in the Ubuntu repository but compile instead?

Either only use packages or compile everything from source would be my recommendation, my personal preferred option is to just compile everything from source.

SDRplay, 2019-02-15:
Don't know if there has ever been a package of SoapySDRPlay - as far as I know it's always needed to be built. Unless anyone else knows different? We have workflows on our website that show the whole build process for SoapySDR, SoapySDRPlay and CubicSDR

Twilight-Logic, 2019-02-15:
SDRplay, when my RSP2 box arrives I will probably install everything on Windows 7 and be done with it. It would be nice to get it working on Mint, but its starting to look like a lot of hassle....

Dantali0n, your points about mixing are well taken and thanks for the further links. I start from the point that for a modern Linux distro, the package ought to be available via apt install. If it is, then it should have all dependencies installed or al least available and work 'out of the box'. In the case of something like SDR, there will be a requirement for drivers, and in this case there isn't one for SDRplay so I guess I will have to remove the apt packages and try and compile everything. Even according to the OS providers, one should always use the distro packages and compiling should always be the last resort. From my experience, compiling rarely goes without problems and one usually has to have a lot of time to spare. When I get the time I will have a go at a full compile. Thanks for your help with this.

Dantali0n, 2019-02-15:
@Twilight-Logic could you be so kind to close the issue.

Cheers,
Dantali0n

Twilight-Logic, 2019-02-15:
I can confirm that having removed the version from the repository followed the workflow instructions on the SDRplay website I have been able to get the API installed and CubicSDR working.
https://www.sdrplay.com/non-windows-workflow/

My RTLSDR seems to work without any additional configuration. I will have to wait for the RSP2 radio to arrive before being testing further with the SDRplay drivers.

closed 2019-02-15 by Twilight-Logic

guruofquality, 2019-02-16:
Thought I would mention that SoapySDR from apt will also look in /usr/local for modules (the default install path for source packages). So its perfectly fine to use the soapysdr from the package manager, but compile new hardware support modules. You can keep your packages installed, but compile from source when something is missing like sdrplay, or perhaps a update or bug fix that you want to try in rtlsdr thats not yet in the package manager, etc. -- just make sure to also install the development files from the package manager. -- https://github.com/pothosware/SoapySDR/wiki/ConfigGuide#when-to-build-from-source