cjcliffe/CubicSDR #708: RumlogNG Integration

tominabox1, 2019-02-19:
(ham radio enhancement) Would be awesome if you could integrate with RumlogNG to get rig control data.  Rumlog provides network accesses to FLdigi and WSJT over network interfaces (specifically uses a MacLoggerDX protocol I believe).

Toontje, 2019-02-27:
i just use a second serial interface to CAT from CubicSDR to the radio and let CubicSDR follow the radio. So when i change the VFO, CubicSDR and RumLog get updated. When i click on a spot in RumLog, the radio changes frequency and updates CubixSDR. When i move the modulator in CubicSDR, the radio changes frequency and updates RumLog.
All works fine.
Instead of working on this, please work on the Notch filter so the automatic scaling doesn't get screwed up by just one spike in the spectrum.

tominabox1, 2019-03-04:
@Toontje Can you explain your implementation? How do you "use a second serial interface"?  My radio only has one serial port

Toontje, 2019-03-28:
Your radio has an RS232 and a USB port and you can use both at the same time.

iz2eer, 2022-05-31:
good morning, i need a help...
i use a ftdx 3000 and rumlongNG..
can you explain, in a easy way, what i have to do to implement a cubicsdr?
i'm not able to organize  the cat chain..
thankyou very much for your time and help