cjcliffe/CubicSDR #721: Option to Disable Visual Features?

ehidle, 2019-03-29:
Hi. I was wondering if there could be an option to disable all visual features (waterfalls, audio scope, etc) to make CubicSDR more friendly for use over remote desktop connections, and to reduce CPU load to an absolute minimum? 

It would be nice to simply be able to say "hey give me a USB receiver at 3.573 MHz and just dump the audio" for unattended use cases. 

I looked in the manual and didn't see anything like this, so thought I would ask.

Thanks to all who work on this program!

vsonnier, 2019-03-31:
This is a duplicate of #577, please see @cjcliffe statement there.

closed 2019-03-31 by vsonnier

ehidle, 2019-03-31:
Hi guys, I wasn't really asking about running without a UI like these other two, but that would certainly be nice. I guess if I were to be very specific, all I'm asking for is maybe an additional option in CMake to disable the main spectrum scope and waterfall. In fact I think my ideal UI setup would be just to keep the demod waterfall turned on and turn off the audio waveform and main spectrum/waterfall. 

Anyway, just wanted to clarify that.

That said, it would be REALLY cool to also be able to launch/run without a UI at all so I can run several instances with several SDRs without bogging down the machine with UI stuff. :-)

starvald, 2019-03-31:
perhaps checkout https://github.com/szpajder/RTLSDR-Airband
am and nfm with soapysdr support.