cjcliffe/CubicSDR #726: SIGSEGV on Fedora 29

mdomsch, 2019-05-07:
(From CubicSDR git commit a7e4d911e92a7063888528afe0efe321c02b9a7b)
Notification time stamped 2019-05-07 12:41:12 UTC

Packages: CubicSDR
Function: __pthread_mutex_lock
First occurrence: 2019-04-13
Type:     core
Count:    10
URL:      https://retrace.fedoraproject.org/faf/reports/bthash/821c4892c6228cdb6e6b47caed58a75b463868c8/

mdomsch, 2019-05-07:
Due to use of wxWidgets 3.1 which is not yet in Fedora <= 30 or rawhide due to it being considered a "development" release, I had backed out the wxWidgets 3.1 changes to use wxWidgets 3.0 since August 2018.  However, AppFrame changes since then make extensive use of wxGLAttributes provided in 3.1 (32 references) which are more difficult to backport.

vsonnier, 2019-05-22:
Hello @mdomsch everyting here is way too old, there is no chance I'm going to consider digging in this problem, sorry.

closed 2019-05-22 by vsonnier