cjcliffe/CubicSDR #727: Scaling and mouse cursor

LightningStalker, 2019-05-07:
For some reason on my system Cubic is not scaling the waterfall images and other GUI elements with the rest of the window. Here is a screenshot: 

[image]

It wouldn't be so bad if it didn't also think the mouse cursor is to the left of where it actually is. As suggested elsewhere here I also installed wxWidgets 3.1.2 but unfortunately it had no effect, unless it requires a reboot or something.

Dantali0n, 2019-05-08:
I think it would help in determining what caused this if you could specify the operating system and how you installed CubicSDR (appimage vs from source)

joelsk, 2019-05-09:
I'm seeing the same thing. I built it on ubuntu 18.10 using the instructions from https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux.

Each individual GL-rendered part of the screen is rendered at half scale but the mouse cursor interaction is happening at full scale within each window.

joelsk, 2019-05-09:
And this happens on my Dell XPS 9560 with both the integrated Intel video hardware or the discrete nvidia hardware.

LightningStalker, 2019-05-13:
OS is Mint using the appimage

joelsk, 2019-05-14:
Just realized that I have a Hidpi (4k) display and I had the ubuntu desktop display scaling set to 200%. I set it back to 100% and, although tiny, the UI of CubicSDR renders correctly and the mouse clicks are in the right places. I suspect the system-rendered bits of the interface respect the scaling but the opengl bits don't.

LightningStalker, 2019-06-06:
The same thing happens on my system. Setting the User interface scaling to anything other than normal will cause this. I think I'm going to stick to Gqrx for now.

ra1nb0w, 2019-09-24:
same trouble on macOS

vsonnier, 2020-04-19:
Hello Linx users, could you please test PR #769 and report if it solve your issues, thanks.

ra1nb0w, 2020-04-19:
you should try https://github.com/cjcliffe/CubicSDR/pull/769 that fix font also (to enable use `-DUSE_OSX_RETINA=1`)

Paul-Nuneaton, 2020-10-09:
So how do I apply a patch?

I downloaded CubicSDR 0.2.6a  from the SDRPlay web site.

I don't want or know how to patch /compile.

Where can I get the executable file from to test?

vsonnier, 2021-08-02:
#769 has been merged, which should fix the problem.

closed 2021-08-02 by vsonnier