cjcliffe/CubicSDR #732: STATUS_STACK_BUFFER_OVERRUN (0xc0000409) in CubicSDR

aolszowka, 2019-05-19:
Hi,

I have been unable to find a PDB that ships with the Release CubicSDR; if you can provide one I can give a better stack trace.

When attempting to add a remote client an unhandled exception is encountered:

### Steps
1. Open CubicSDR
2. Click on Remote
3. Click Add
4. Select rtlsdr
5. addr=127.0.0.1:1234
6. Click Ok

This will then error

### Environment
* Windows 10 Pro 1809 x64
* rtl_tcp running on localhost port 1234
* nesdr smart v3

### Video
[image]

### Exception
Unhandled exception at 0x00007FFF47D8E91E (ucrtbase.dll) in CubicSDR.exe: Fatal program exit requested.

### Stack Trace
>	ucrtbase.dll!abort()	Unknown
 	ucrtbase.dll!terminate()	Unknown
 	vcruntime140.dll!00007fff379c2330()	Unknown
 	vcruntime140.dll!00007fff379c1e40()	Unknown
 	vcruntime140.dll!00007fff379cba82()	Unknown
 	ntdll.dll!00007fff4ac9477f()	Unknown
 	ntdll.dll!00007fff4abf4bef()	Unknown
 	ntdll.dll!00007fff4ac934ee()	Unknown
 	KernelBase.dll!00007fff46f89129()	Unknown
 	vcruntime140.dll!00007fff379c2c6f()	Unknown
 	ntdll.dll!00007fff4ac93cc6()	Unknown
 	CubicSDR.exe!00007ff788830a49()	Unknown
 	CubicSDR.exe!00007ff788804225()	Unknown
 	CubicSDR.exe!00007ff788803ac9()	Unknown
 	ucrtbase.dll!thread_start<unsigned int (__cdecl*)(void * __ptr64)>()	Unknown
 	kernel32.dll!00007fff486b7974()	Unknown
 	ntdll.dll!00007fff4ac5a271()	Unknown

jazzkutya, 2019-05-20:
Hi,

that remote in cubicsdr is not rtl_tcp remote but SoapySDRRemote. See https://github.com/pothosware/SoapySDR/wiki#remote-access and https://github.com/pothosware/SoapyRemote/wiki . Run SoapySDRServer instead of rtl_tcp.

vsonnier, 2019-05-22:
@jazzkutya is right, you need to use the matching `SoapySDRServer`, not `rtl_tcp`. If you don't build it yourself from https://github.com/pothosware/SoapyRemote, you can find pre-built binaries in the Win64 PothosSDR distribution: https://downloads.myriadrf.org/builds/PothosSDR/.

The matching command would be, in your exemple: `SoapySDRServer.exe --bind=127.0.0.1:1234`

Now if your goal is to interface the `rtl_xxx` familly of tools with CubicSDR, this is simply not possible, the protocols are different between `SoapySDRRemote` and `rtl_tcp` and such.

vsonnier, 2019-05-22:
BTW, your config is similar to mine and CubicSDR works for me, obviously :)

vsonnier, 2019-06-01:
@aolszowka Any news ?

aolszowka, 2019-06-02:
Hey sorry about that; I was attempting to use rtl_tcp only because I wanted to also use the same radio in conjunction with rtlamr (https://github.com/bemasher/rtlamr) to read my electric and water until I could pick up additional radios on my Pi. I will close this issue as resolved.

I realize this is not a support forum but any recommendations on how best to do this? Long term I was going to have multiple radios on a single Raspberry Pi 3 B+ and have them act as a network attached SDR's for consumption by other consumers on the network. I guess what I could do is have some radios run SoapySDRServer and others run rtl_tcp?

Thank you for the project I have loved being able to listen to FM radio via it :).

closed 2019-06-02 by aolszowka

vsonnier, 2019-06-02:
What you need is several `rtl_tcp` runing, each one using a different RTL-SDR. The problem is all RTL-SDR have the same serial number /id by default, so they can't be be taken appart from each other if they are plugged to the same device. 
To solve that, you have to force each of them to have a different SN/Id by using `rtl_eeprom` on them. Look for Internet for further details.