cjcliffe/CubicSDR #734: Not Found (insert SDR name here)

TheButterZone, 2019-05-21:
[image]

Are they lying, or what exact format is required to get CubicSDR to connect to the green AirSpy or RTL-SDR servers on https://airspy.com/directory/# ?

jazzkutya, 2019-05-21:
Hi,

they are not lying. Airspy servers work only with sdr# while the airspy device itself should work with cubicsdr (locally, not over the network). Those servers use an sdr#-specific protocol. Cubicsdr can use devices over the network with SoapySDRRemote https://github.com/pothosware/SoapyRemote/wiki but this is a different solution for a similar (not same) problem and they are not compatible.

TheButterZone, 2019-05-21:
<SA

> 6BSS-Mike> thats for airspy HW, not the spy server network
> <mzs> *agree SA6BSS-Mike* I just searched through the source code for cubicsdr on github and no it does not support spyservers. It could be added if they used the code from https://github.com/opensatelliteproject/xritdemod/tree/master/demodulator/src

vsonnier, 2019-05-22:
Hello there, @jazzkutya is right CubicSDR is compatible only with SoapySDRRemote because it allows interfacing with distant hardware the same way as local hardware using the SoapySDR API. The thing is, CubicSDR usage of this API is deeply interlaced, and I don't see it using "yet another API".

On the SoapySDR side itself SpyServer support has already been suggested for instance here https://github.com/pothosware/SoapySDR/issues/156 and suggestions may come there. It is indeed better and easier to make SoapySDRRemote or why not a new "SoapySDRSpyServer" module support the new devices than plugin another code in the client CubicSDR side of things.

closed 2019-05-22 by vsonnier