cjcliffe/CubicSDR #736: RTL-SDR Access issue (v0.2.5)

dmitryelj, 2019-05-22:
Hi,
Thanks for the cool product.

In the last version I've found a bug: in CubicSDR manual gain does not work for RTL-SDR, and the serial number field is empty. 0.2.4 works ok (see screenshots).
System Windows 10, last version was taken from PothosSDR-2019.03.24-vc14-x64.exe.

0.2.4:
[image]

0.2.5:
[image]

_SoapySDRUtil --probe="driver=rtlsdr"_ command returns correct gains, so SoapySDR itself is working.

vsonnier, 2019-05-25:
Hello @dmitryelj, I also tried the PothosSDR distribution and I'm running Windows 10 also.
I tried both their CubicSDR and my compiled CubicSDR `master` with a RTL-SDR.com v3 dongle.

One time I got your problem showing a half-recognized dongle with `Generic RTL2832U OEM ::` at first but couldn't reproduce it afterwards, all the times it was OK.

Digging old issues, I see that we had problems in the past making Nooelec being recognized properly. Another user reported having fix its issues with removing the CubicSDR configuration:

In Explorer, go here with the adress bar: `%APPDATA%\CubicSDR`, and it should open someting like `C:\Users\[your account name]\AppData\Roaming\CubicSDR` and remove the `config.xml` then restart.

dmitryelj, 2019-05-25:
Thanks for the answer.

I removed the file and tried again, the result is the same. 
Interesting fact is that the changes (sampe rate, AGC mode) are not saving after the exit.

Tried also to run the app "as admin", no changes.

dmitryelj, 2019-05-25:
Lo attached:
[start.log](https://github.com/cjcliffe/CubicSDR/files/3219232/start.log)

vsonnier, 2019-05-25:
Yep, your device is not detected properly. Try to install the very latest Zadig drivers (v2.4) found here: https://zadig.akeo.ie/.

Here the problem lies with the SoapySDR module, not with CubicSDR v0.24 vs. master differences I'm afraid.

What is your RTL-SDR dongle origin ?

dmitryelj, 2019-05-25:
Its looks weird because SoapySDRUtil from the same folder works good.
[image]

I use RTL SDR V3 as well.

dmitryelj, 2019-05-25:
Small addition: I tried Python client for SoapySDR - the result is really random, sometimes it works and sometimes its not. Looks like its really not the CubicSDR issue.

dmitryelj, 2019-05-25:
I created a ticket at SoapyRTLSDR page: https://github.com/pothosware/SoapyRTLSDR/issues/38

closed 2019-05-25 by dmitryelj

vsonnier, 2019-05-25:
Thanks @dmitryelj your ticket looks very extensive, great !