cjcliffe/CubicSDR #738: Application termination sequence fails

marjinal1st, 2019-05-24:
I've compiled the latest (0fa818c) CubicSDR on my machine, using:

* Ubuntu 18.04.2 x86_64 (4.18.0-20-generic)
* gcc 7.4.0
* All of the dependencies in the build instructions

Application starts and works properly, but when I close it I get this error:

```
Loaded font 'Bitstream Vera Sans Mono' from '/home/marjinal1st/Development/SDR/CubicSDR-Release/x64/fonts/vera_sans_mono27_0.png', parsed 255 characters.
Terminating SDR thread..
ERROR: thread '9SDRThread' has not terminated in time ! (> 3000 ms)
```

I've debugged it and found the actual source of the problem:

* IOThread::terminate method is invoked
* **stopping** atomic boolean is set true
* IOThread::isTerminated method is invoked
* **terminated** atomic boolean is checked if it's true, but it's not.
* Thus, the method never returns true and termination sequence fails.

### Possible Fixes

Setting **terminated** true, when IOThread::terminate is invoked:

```cpp
void IOThread::terminate() {
    terminated.store(true);
    stopping.store(true);
};
```

Or checking if **stopped** is true, when IOThread::isTerminated is invoked:

```cpp
bool IOThread::isTerminated(int waitMs) {

    if (terminated.load() || stopping.load()) {
        return true;
    }
    else if (waitMs == 0) {
        return false;
    }

    //this is a stupid busy plus sleep loop
    int nbCyclesToWait = 0;

    if (waitMs < 0) {
        nbCyclesToWait = std::numeric_limits<int>::max();
    }
    else {

        nbCyclesToWait = (waitMs / SPIN_WAIT_SLEEP_MS) + 1;
    }

    for ( int i = 0; i < nbCyclesToWait; i++) {

        std::this_thread::sleep_for(std::chrono::milliseconds(SPIN_WAIT_SLEEP_MS));

        if (terminated.load()) {
            return true;
        }
    }

    std::cout << "ERROR: thread '" << typeid(*this).name() << "' has not terminated in time ! (> " << waitMs << " ms)" << std::endl << std::flush;

    return terminated.load();
}
```

Both of the solutions seems to work and application terminates w/o any problem:

```
Loaded font 'Bitstream Vera Sans Mono' from '/home/marjinal1st/Development/SDR/CubicSDR-Release/x64/fonts/vera_sans_mono27_0.png', parsed 255 characters.
Terminating SDR thread..
Terminating SDR post-processing thread..
Terminating All Demodulators..
Terminating Visual Processor threads..
Application termination complete.
```

I didn't open any PR because I'm not sure if these fixes are applicable.

vsonnier, 2019-05-25:
Hello @marjinal1st, as far as I see what you observe is intended, and properly described in `IOThread.h`:
- `terminate()` is a request for termination setting the `stopping` to true,
- `terminated` is only set to true when the actual termination occurs at the end of thread entry point function return, either by normal flow or exception,
- only then `isTerminated` returns true.

So, the proposed fixes are all wrong I'm afraid.

Now the `ERROR: thread '9SDRThread' has not terminated in time ! (> 3000 ms)` error indeed pop up if you start CubicSDR and just close it without starting a device. I admit I never bother to workaround this particular edge case.

If your case is different, it may reveal a problem elswhere, but the `IOThread` termination procedure is right. The goal is precisely not to declare a thread stopped just because is has been requested to, but to properly wait for actual termination in order to proceed to safe resource reclamation.

marjinal1st, 2019-05-25:
> So, the proposed fixes are all wrong I'm afraid.

That's what I suspected :smile: 

Also, you're right about my case, I didn't start any device and tried to quit. Just to be sure, plugged in a regular RTLSDR dongle, started it and closed the application properly.

Still, I debugged deeper and found this:

* When no device is started, `SDRThread` object's `threadMain` method is never called.
* So `terminated` is never set true, thus `isTerminated` return false.

This means the actual problem is in the `CubicSDR::OnExit`. This method is trying to terminate and check the SDR thread, but that thread is not actually running. I think it should be aware of whether thread is started or not :thinking:

vsonnier, 2019-05-25:
As I said, an edge case I didn't bother to code for. PR is welcome, or else I'll close this issue.

vsonnier, 2019-06-02:
Timeout. Since I'm not going to to it, lets wontfix this.

closed 2019-06-02 by vsonnier