cjcliffe/CubicSDR #740: Windows MSVC 2019 requires _HAS_STD_BYTE=0

MarcKaufman, 2019-06-03:
The latest version of MSVC 2019 need to have the preprocessor symbol _HAS_STD_BYTE=0 in order to prevent ambiguous definitions of "byte"

vsonnier, 2019-06-03:
Hello @MarcKaufman, you are apparently forcing C++ standard to C++17, which is way too advanced. Following the compiling guide and using CMake to generate the VS solution works flawlessly, and this is the only supported way.
From that, you are free to tinker the VS solution further, but we are not going to offer support for that, sorry.

closed 2019-06-03 by vsonnier

MarcKaufman, 2019-06-03:
Yes, I always push my development tools to the latest revision. MSVC 15 won't be around forever. You'll need to do this, someday.