cjcliffe/CubicSDR #742: CubicSDR Windows 32/64 bit test builds (0.2.6 alpha 1)

cjcliffe, 2019-07-01:
Ahoy,

Here's a couple test builds using the latest Visual Studio 2019; would like to see if everything is still installing and working as intended; Windows users please give them a try and let me know if anything unexpected occurs.

This build includes updated SoapySDR modules and libraries for most devices and also adds support for PlutoSDR.

[CubicSDR-0.2.6.a1-win32.exe.zip](https://github.com/cjcliffe/CubicSDR/files/3346711/CubicSDR-0.2.6.a1-win32.exe.zip)
[CubicSDR-0.2.6.a1-win64.exe.zip](https://github.com/cjcliffe/CubicSDR/files/3346714/CubicSDR-0.2.6.a1-win64.exe.zip)

magellan-13016, 2019-07-02:
Many thanks. I will test the gift next Thursday and give you feedback. Now all students are in holiday but I will use it for the next educational project from September. 

Many many thanks for not letting me down.

vsonnier, 2019-08-17:
Hello Charles @cjcliffe , sorry for playing dead I didn't have much time lately. After quick tests, it look OK on the Windows side but I don't know when I can get more time to test in-depth.

vsonnier, 2019-09-19:
Okaay, first bugs : 
- Double-click on Recent entry do not re-align WF on the new frequency if it is not in range. 
- The letters [S/M] for instance are not displayed simultaneously on a S and M demod.

vsonnier, 2019-10-29:
> Double-click on Recent entry (often) do not re-align WF on the new frequency if it is not in range.

Definitively wrong here, the WF/Spectrum synchro on the target range works or not randomly. I'll try to investigate before 2019 is over, but no deadline. 
In the meantime, wxWidgets 3.1.3 is out, works great with VS2019 (as well as before, that is) so I'm updating the Build Windows Wiki accordingly.

vsonnier, 2019-12-18:
Hi @cjcliffe, I wish you a Merry Christmas. Alas Santa-Claus do not exist (sorry) and I won't have time and I confess, motivation, to fix the reported issues in time. 

Since you already invest a lot of time in refactoring in the past months, I think we can consider this version good to go, when you see it ready for release.

cjcliffe, 2019-12-31:
@vsonnier no worries; Merry Christmas and a Happy New Year! -- I should be around more often in the next few months.

vsonnier, 2019-12-31:
Hi ! Now that RSPDuo support is on the way : https://github.com/pothosware/SoapySDRPlay/issues/62
with some adaptations to CubicSDR I have yet to review by @fventuri related to a cleaner switch of `Devices` at runtime, we better NOT release Cubic right now :)

closed 2020-06-17 by cjcliffe