cjcliffe/CubicSDR #744: CubicSDR won't run on Mac

soakland5, 2019-07-18:
I just got an SDRPlay RSP1 for my Mac and loaded both of the Mac software files from the sdrplay website and it says "no soapy modules installed". What may I be doing wrong?

SDRplay, 2019-07-18:
what does SoapySDRUtil --info report? I'm assuming you installed both the API and CubicSDR?

soakland5, 2019-07-18:
I can't find that utility, is it a seperate download or is it in my
directories som

SDRplay, 2019-07-18:
if you have installed the API, then in a terminal window you should be able to type...

SoapySDRUtil --info

soakland5, 2019-07-18:
Can't get to the directory. I see now I have the program running I think.
I don't have the cables ready yet so can't try it but I have a display. I
still don't understand the error message

SDRplay, 2019-07-18:
Not sure I follow - you have to have the RSP connected for any of this to work. Is it not connected? I'm not sure what directory you are trying to get to - the terminal window is an application that allows you to type commands in. If you type in SoapySDRUtil --info, this will tell you what SoapySDR modules are loaded.

ra1nb0w, 2019-07-27:
`libmirsdrapi-rsp.dylib` is not distributable with the package; only for personal usage.
See [CMakeLists.txt](https://github.com/cjcliffe/CubicSDR/blob/master/CMakeLists.txt#L806).

Install [macports](https://www.macports.org/install.php) and then install the packages with
```
sudo port install cubicsdr SoapySDRPlay
```
Now you have the latest commit (I am sending the update right now).
Sometimes you need to run `SoapySDRUtil --info` or refresh on CubicSDR two times to discovery the device.

Dantali0n, 2020-01-09:
> Another generic issue 'Not working on my Mac' with no feedback for months. Thanks @ra1nb0w and @SDRplay for trying to help, but I'm closing this because it get too old.

Maybe we could define some form of policy that we include in the 'submit issue description' (configurable from settings) that explains exactly after how long we will close stale / unresponsive issues.