cjcliffe/CubicSDR #746: Problems compiling on Slackware64-current

pchristyUK, 2019-08-16:


pchristyUK, 2019-08-16:
System: Slackware64-current with Plasma 5 upgrade.
Compilation of 0.2.5 fails with:
`/tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.cpp: In constructor ‘ModemAM::ModemAM()’:
/tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.cpp:7:36: error: cannot convert ‘double’ to ‘liquid_ampmodem_type’
    7 |     demodAM = ampmodem_create(0.5, 0.0, LIQUID_AMPMODEM_DSB, 0);
      |                                    ^~~
      |                                    |
      |                                    double
In file included from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/Modem.h:6,
                 from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.h:5,
                 from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.cpp:4:
/usr/include/liquid/liquid.h:7242:47: note:   initializing argument 2 of ‘ampmodem_s* ampmodem_create(float, liquid_ampmodem_type, int)’
 7242 |                          liquid_ampmodem_type _type,
      |                          ~~~~~~~~~~~~~~~~~~~~~^~~~~
In file included from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/audio/AudioThread.h:13,
                 from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/Modem.h:8,
                 from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemDSB.h:5,
                 from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemDSB.cpp:4:
/tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/external/rtaudio/RtAudio.h: In copy constructor ‘S24::S24(const S24&)’:
/tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/external/rtaudio/RtAudio.h:654:33: warning: implicitly-declared ‘S24& S24::operator=(const S24&)’ is deprecated [-Wdeprecated-copy]
  654 |   S24( const S24& v ) { *this = v; }
      |                                 ^
/tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/external/rtaudio/RtAudio.h:654:3: note: because ‘S24’ has user-provided ‘S24::S24(const S24&)’
  654 |   S24( const S24& v ) { *this = v; }
      |   ^~~
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:323: CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemDSB.cpp: In constructor ‘ModemDSB::ModemDSB()’:
/tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemDSB.cpp:7:40: error: cannot convert ‘double’ to ‘liquid_ampmodem_type’
    7 |     demodAM_DSB = ampmodem_create(0.5, 0.0, LIQUID_AMPMODEM_DSB, 1);
      |                                        ^~~
      |                                        |
      |                                        double
In file included from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/Modem.h:6,
                 from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemDSB.h:5,
                 from /tmp/SBo/CubicSDR-0.2.5/CubicSDR-0.2.5/src/modules/modem/analog/ModemDSB.cpp:4:
/usr/include/liquid/liquid.h:7242:47: note:   initializing argument 2 of ‘ampmodem_s* ampmodem_create(float, liquid_ampmodem_type, int)’
 7242 |                          liquid_ampmodem_type _type,
      |                          ~~~~~~~~~~~~~~~~~~~~~^~~~~
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:336: CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemDSB.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
`
Any ideas?
-- 
Pete

vsonnier, 2019-08-17:
Your liquid-dsp is not up-to-date w.r.t the `ampmodem` API changes done here : https://github.com/cjcliffe/CubicSDR/pull/724. To be clear, update liquid-dsp to the recently released v1.32 version and it shoukd be fine.

pchristyUK, 2019-08-18:
Thank you for the reply! I am already running liquid-sdr-1.3.2, and trying to build CubicSDR-0.2.5. CubicSDR-0.2.4 produces the same error.

-- 
Pete

vsonnier, 2019-08-18:
> Thank you for the reply! I am already running liquid-sdr-1.3.2, and trying to build CubicSDR-0.2.5.

That is the problem. CubicSDR 0.25 `tag` is way too old. Either you compile CubicSDR `master` from source, or you try 0.25 using liquid-dsp 1.3.1.

pchristyUK, 2019-08-18:
OK! I thought that 0.2.5 was the latest stable versions. Downloading git now...

pchristyUK, 2019-08-18:
YES! The git version worked! Many thanks!
Might I respectfully suggest issuing a newer release - even if its only an interim version. I've had issues with git versions of some software in the past, and try and use release versions where possible.

-- 
Pete

vsonnier, 2019-08-19:
> Might I respectfully suggest issuing a newer release - even if its only an interim version.

0.26 in on the works, but don't hold your breath it may take 3 days or 3 months to be released, so please continue using `master`.

> I've had issues with git versions of some software in the past, and try and use release versions where possible.

For CubicSDR the `master` branch is always the best version :) The tagged versions are merely snapshots of the current master when @cjcliffe see fit to release a new version.

Anyway, have fun with CubicSDR !

closed 2019-08-19 by vsonnier

Geremia, 2020-04-03:
There's this issue compiling d2f9333 on Slackware 14.2 with liquid-dsp 1.3.2:
```
/tmp/SBo/CubicSDR-d2f9333/CubicSDR-d2f9333/src/modules/modem/analog/ModemAM.cpp: In constructor 'ModemAM::ModemAM()':
/tmp/SBo/CubicSDR-d2f9333/CubicSDR-d2f9333/src/modules/modem/analog/ModemAM.cpp:7:58: error: invalid conversion from 'int' to 'liquid_ampmodem_type' [-fpermissive]
     demodAM = ampmodem_create(0.5, LIQUID_AMPMODEM_DSB, 0);
                                                          ^
/tmp/SBo/CubicSDR-d2f9333/CubicSDR-d2f9333/src/modules/modem/analog/ModemAM.cpp:7:58: error: too few arguments to function 'ampmodem_s* ampmodem_create(float, float, liquid_ampmodem_type, int)'
In file included from /tmp/SBo/CubicSDR-d2f9333/CubicSDR-d2f9333/src/modules/modem/Modem.h:6:0,
                 from /tmp/SBo/CubicSDR-d2f9333/CubicSDR-d2f9333/src/modules/modem/analog/ModemAM.h:5,
                 from /tmp/SBo/CubicSDR-d2f9333/CubicSDR-d2f9333/src/modules/modem/analog/ModemAM.cpp:4:
/usr/include/liquid/liquid.h:5717:10: note: declared here
 ampmodem ampmodem_create(float _m,
          ^
CMakeFiles/CubicSDR.dir/build.make:566: recipe for target 'CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o' failed
make[2]: *** [CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o] Error 1
```
https://pastebin.com/zexDagBR

Dantali0n, 2020-04-03:
> @vsonnier There's this issue compiling [d2f9333](https://github.com/cjcliffe/CubicSDR/commit/d2f93335236ef37557057d7b71c46969bec14471) on Slackware 14.2 with liquid-dsp 1.3.2:
> 

This patch has been merged with master so instead of checking out the specific commit mentioned I recommend checking out the latest master and compiling that instead.

Geremia, 2020-04-03:
@Dantali0n d2f9333 is not the latest master?

Dantali0n, 2020-04-04:
> @Dantali0n [d2f9333](https://github.com/cjcliffe/CubicSDR/commit/d2f93335236ef37557057d7b71c46969bec14471) is not the latest master?

Apparently it is the lastest master.

I am almost certain you are still using liquid-dsp 1.3.**1** as your headers clearly indicate they expect the signature `ampmodem_create(float, float, liquid_ampmodem_type, int)`. This was changed from 1.3.1 to 1.3.2 in `ampmodem_create(float, liquid_ampmodem_type, int)`.

See also: https://github.com/jgaeddert/liquid-dsp/blob/v1.3.2/src/modem/src/ampmodem.c#L66

Geremia, 2020-04-11:
@Dantali0n _Mea culpa._ I had liquid-dsp 1.3.0 installed. It works fine now with 1.3.2.