cjcliffe/CubicSDR #749: Issue when Compiling

w-wright, 2019-08-31:
Hello,

I have just been trying to complile CubicSDR and came across this issue:

`/home/wew/Dev/CubicSDR-0.2.4/src/modules/modem/analog/ModemAM.cpp:7:63: error: cannot convert ‘double’ to ‘liquid_ampmodem_type’
     demodAM = ampmodem_create(0.5, 0.0, LIQUID_AMPMODEM_DSB, 0);
                                                               ^
In file included from /home/wew/Dev/CubicSDR-0.2.4/src/modules/modem/Modem.h:6,
                 from /home/wew/Dev/CubicSDR-0.2.4/src/modules/modem/analog/ModemAM.h:5,
                 from /home/wew/Dev/CubicSDR-0.2.4/src/modules/modem/analog/ModemAM.cpp:4:
`

Any ideas?

Dantali0n, 2019-08-31:
I ran into the same issue recently, you need to update the version of liquid-dsp by updating the liquid-dsp git repository and compiling + installing it again.

vsonnier, 2019-08-31:
Indeed, this is a duplicate of #746.

closed 2019-08-31 by vsonnier