cjcliffe/CubicSDR #750: Disable audio

aik, 2019-09-05:
Is there a way to disable audio in linux?

parkerlreed, 2019-09-05:
As in processing or just mute it?

aik, 2019-09-05:
mute. there is no option "no audio" so I select one of those which are not connected anywhere but every time I tune to a new frequency, cubucsdr switches to the default audio and turns that on.
I am trying to see activities in wifi range so listening to this does not make sense.

parkerlreed, 2019-09-06:
The button right below the volume slider on the far top right should be mute.

parkerlreed, 2019-09-06:
[image]

vsonnier, 2019-09-06:
@parkerlreed As @parkerlreed says : `The M (Mute)` button mute all. Note that you can mute a demod individually by pressing `M` on a currently highlighted demod.

closed 2019-09-06 by vsonnier