cjcliffe/CubicSDR #752: Error building module ModemAM.cpp on  xubuntu 18.04, cannot convert ‘double’ to ‘liquid_ampmodem_type

alfargan, 2019-09-07:
Hi. I have  problems when  try to compile cubicsdr (v.0.2.4), I have always the same error building  ModemAM.cpp ( see below ):

kavier@cygnus:~/Escritorio/CubicSDR-0.2.4/build$ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
[  3%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppFrame.cpp.o
/home/kavier/Escritorio/CubicSDR-0.2.4/src/AppFrame.cpp: In member function ‘bool AppFrame::actionOnMenuSettings(wxCommandEvent&)’:
/home/kavier/Escritorio/CubicSDR-0.2.4/src/AppFrame.cpp:1286:62: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (event.GetId() >= wxID_ANTENNAS_BASE && event.GetId() < wxID_ANTENNAS_BASE + antennaNames.size()) {
                                                ~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/kavier/Escritorio/CubicSDR-0.2.4/src/AppFrame.cpp: In member function ‘bool AppFrame::actionOnMenuRig(wxCommandEvent&)’:
/home/kavier/Escritorio/CubicSDR-0.2.4/src/AppFrame.cpp:1675:48: warning: unused parameter ‘event’ [-Wunused-parameter]
 bool AppFrame::actionOnMenuRig(wxCommandEvent& event) {
                                                ^~~~~
/home/kavier/Escritorio/CubicSDR-0.2.4/src/AppFrame.cpp: In member function ‘void AppFrame::OnAboutDialogClose(wxCommandEvent&)’:
/home/kavier/Escritorio/CubicSDR-0.2.4/src/AppFrame.cpp:2371:51: warning: unused parameter ‘event’ [-Wunused-parameter]
 void AppFrame::OnAboutDialogClose(wxCommandEvent& event) {
                                                   ^~~~~
[  4%] Building CXX object CMakeFiles/CubicSDR.dir/src/AppConfig.cpp.o
[  5%] Building CXX object CMakeFiles/CubicSDR.dir/src/FrequencyDialog.cpp.o
[  7%] Building CXX object CMakeFiles/CubicSDR.dir/src/DemodLabelDialog.cpp.o
[  8%] Building CXX object CMakeFiles/CubicSDR.dir/src/IOThread.cpp.o
[  9%] Building CXX object CMakeFiles/CubicSDR.dir/src/ModemProperties.cpp.o
/home/kavier/Escritorio/CubicSDR-0.2.4/src/ModemProperties.cpp: In member function ‘void ModemProperties::initDefaultProperties()’:
/home/kavier/Escritorio/CubicSDR-0.2.4/src/ModemProperties.cpp:76:13: warning: variable ‘defaultDevice’ set but not used [-Wunused-but-set-variable]
         int defaultDevice = 0;
             ^~~~~~~~~~~~~
/home/kavier/Escritorio/CubicSDR-0.2.4/src/ModemProperties.cpp: In member function ‘void ModemProperties::OnCollapse(wxPropertyGridEvent&)’:
/home/kavier/Escritorio/CubicSDR-0.2.4/src/ModemProperties.cpp:264:55: warning: unused parameter ‘event’ [-Wunused-parameter]
 void ModemProperties::OnCollapse(wxPropertyGridEvent &event) {
                                                       ^~~~~
/home/kavier/Escritorio/CubicSDR-0.2.4/src/ModemProperties.cpp: In member function ‘void ModemProperties::OnExpand(wxPropertyGridEvent&)’:
/home/kavier/Escritorio/CubicSDR-0.2.4/src/ModemProperties.cpp:268:53: warning: unused parameter ‘event’ [-Wunused-parameter]
 void ModemProperties::OnExpand(wxPropertyGridEvent &event) {
                                                     ^~~~~
[ 10%] Building CXX object CMakeFiles/CubicSDR.dir/src/BookmarkMgr.cpp.o
[ 11%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRDeviceInfo.cpp.o
[ 12%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDRPostThread.cpp.o
[ 14%] Building CXX object CMakeFiles/CubicSDR.dir/src/sdr/SDREnumerator.cpp.o
[ 15%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorPreThread.cpp.o
[ 16%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorThread.cpp.o
[ 17%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorWorkerThread.cpp.o
[ 18%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorInstance.cpp.o
[ 20%] Building CXX object CMakeFiles/CubicSDR.dir/src/demod/DemodulatorMgr.cpp.o
[ 21%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/Modem.cpp.o
[ 22%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/ModemAnalog.cpp.o
[ 23%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/ModemDigital.cpp.o
[ 24%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o
/home/kavier/Escritorio/CubicSDR-0.2.4/src/modules/modem/analog/ModemAM.cpp: In constructor ‘ModemAM::ModemAM()’:
/home/kavier/Escritorio/CubicSDR-0.2.4/src/modules/modem/analog/ModemAM.cpp:7:63: error: cannot convert ‘double’ to ‘liquid_ampmodem_type’ for argument ‘2’ to ‘ampmodem_s* ampmodem_create(float, liquid_ampmodem_type, int)’
     demodAM = ampmodem_create(0.5, 0.0, LIQUID_AMPMODEM_DSB, 0);
                                                               ^
CMakeFiles/CubicSDR.dir/build.make:542: recipe for target 'CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o' failed
make[2]: *** [CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o] Error 1
CMakeFiles/Makefile2:99: recipe for target 'CMakeFiles/CubicSDR.dir/all' failed
make[1]: *** [CMakeFiles/CubicSDR.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

¿Some ideas?

Thanks

alfargan, 2019-09-07:
Sorry.... Duplicate. I apologize

vsonnier, 2019-09-07:
Indeed, a duplicate of #746, #749, #751.

closed 2019-09-07 by vsonnier