cjcliffe/CubicSDR #754: Support for ALSA sub-devices?

ab1jx, 2019-09-09:
I'm on an Odroid N2 which has a soundcard device on the motherboard.  But it can do sound to HDMI, analog, or SPDIF so 3 subdevices on one "card" to use the ALSA term.  "aplay -l" looks like this:

**** List of PLAYBACK Hardware Devices ****
card 0: AMLAUGESOUND [AML-AUGESOUND], device 0: SPDIF-B-dit-hifi dit-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: AMLAUGESOUND [AML-AUGESOUND], device 1: TDM-C-T9015-audio-hifi T9015-audio-hifi-1 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 0: AMLAUGESOUND [AML-AUGESOUND], device 2: SPDIF-dit-hifi dit-hifi-2 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

On card 0 there are 3 devices 0-2.  I built CubicSDR from a git clone, it defaulted to PulseAudio.  There was one device in the audio device selection which didn't work.  So I uninstalled and selected ALSA in the cmake cache, rebuilt and reinstalled.  Now I don't have an audio device selection menu item at all, and still no sound from CubicSDR but it works in other things like smplayer.  My /etc/asound.conf looks like:

# from https://wiki.odroid.com/odroid-n2/application_note/sound
pcm.!default {
        type plug
        slave {
        pcm "hw:0,1"
        }
}

This is set for the analog output (hdmi is default).  So this defaults to analog because of the "hw:0,1" (card 0 device 1) but can I make this work with CubicSDR?

parkerlreed, 2019-09-10:
If you go the Pulseaudio route, are you not able to switch the device?

With pulseaudio and pulseaudio-alsa installed

An easy way to check is Pavucontrol on the configuration tab

[image]

You should be able to switch outputs in realtime and have all audio go the route you want.

ab1jx, 2019-09-10:
I didn't have RtAudio installed, then in the cmake cache I read:
//Use the system RtAudio which in turn provides OSS, ALSA, JACK,
// PulseAudio support depending on how it was compiled

So I installed that and Jack Audio.  Sometimes I get sound and sometimes not, sometimes a queue fills up.  The audio device choices seem to depend on the modulation.  I don't have a choice except at certain times.   That audio out section doesn't appear until I click on a signal, there's a pop, then it shows up.

When I tried PulseAudio it killed all my sound from every program until I uninstalled it.  Maybe everything was getting routed through it.  At least it's not dead anymore.
[image]

vsonnier, 2019-09-10:
> sometimes a queue fills up.

Your device has not enough CPU power, so try to reduce the sample rate and only create one demod.

>  That audio out section doesn't appear until I click on a signal, there's a pop, then it shows up.

Perfectly normal, this is the way to create a demod.

Please see the manual here for concepts and basic handling : https://cubicsdr.readthedocs.io/en/latest/

ab1jx, 2019-09-10:
I only had one demod I think.  The CPU usage setting in the menu (and watching top) seems to always put me at 200-300% CPU (2-3 of 6 cores).  But if it's doing graphics without using OpenGL ES it's not taking advantage of the Mali at all.  Peter Onion at raspberrypi.org did an SDR where the waterfall was an OpenGL texture that rotated so for each spectrum trace it just needed to be drawn onto the edge of the texture and the GPU would handle scrolling it down.  There's OpenGL stuff in the CMakeCache.txt but not much in the build instructions https://github.com/cjcliffe/CubicSDR/wiki/Build-Linux about how to set it up.  I don't want software OpenGL but it works for getting it going initially.  I haven't progressed to the point of writing OpenGL but I can appreciate what it can do.  I don't have desktop machines anymore, just these little ARM boxes.  https://www.hardkernel.com/shop/odroid-n2-with-4gbyte-ram/

ab1jx, 2019-09-11:
Here's an example of something not quite right.  I looked up a frequency I wanted to monitor: 935.450 MHz.  Brought up CubicSDR and typed in the frequency and hit enter.  Not much happened.  I don't hear anything, I selected nbfm afterward but I don't see the audio device properties stuff.  It registers as a "center" frequency, but what is that really, the local oscillator frequency?  Not very useful.

I finally got it by clicking in the waterfall, adjusting each digit of the frequency, and that started a demodulator I guess so I could go into that and choose the audio device.  It would be more useful to accept the typed-in frequency as the frequency of interest (tuned frequency) and put the local oscillator (center) frequency a little above or below it.

[image]

closed 2019-09-12 by ab1jx