cjcliffe/CubicSDR #755: Cubic 0.2.5 will not build.

masterplayer30, 2019-09-17:
Here is the CLI output:

[ 23%] Building CXX object CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o
In file included from /home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/src/audio/AudioThread.h:13,
                 from /home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/src/modules/modem/Modem.h:8,
                 from /home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.h:5,
                 from /home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.cpp:4:
/home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/external/rtaudio/RtAudio.h: In copy constructor ‘S24::S24(const S24&)’:
/home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/external/rtaudio/RtAudio.h:654:33: warning: implicitly-declared ‘S24& S24::operator=(const S24&)’ is deprecated [-Wdeprecated-copy]
  654 |   S24( const S24& v ) { *this = v; }
      |                                 ^
/home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/external/rtaudio/RtAudio.h:654:3: note: because ‘S24’ has user-provided ‘S24::S24(const S24&)’
  654 |   S24( const S24& v ) { *this = v; }
      |   ^~~
/home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.cpp: In constructor ‘ModemAM::ModemAM()’:
/home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.cpp:7:36: error: cannot convert ‘double’ to ‘liquid_ampmodem_type’
    7 |     demodAM = ampmodem_create(0.5, 0.0, LIQUID_AMPMODEM_DSB, 0);
      |                                    ^~~
      |                                    |
      |                                    double
In file included from /home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/src/modules/modem/Modem.h:6,
                 from /home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.h:5,
                 from /home/bobby/.cache/yay/cubicsdr/src/CubicSDR-0.2.5/src/modules/modem/analog/ModemAM.cpp:4:
/usr/include/liquid/liquid.h:7242:47: note:   initializing argument 2 of ‘ampmodem_s* ampmodem_create(float, liquid_ampmodem_type, int)’
 7242 |                          liquid_ampmodem_type _type,
      |                          ~~~~~~~~~~~~~~~~~~~~~^~~~~
make[2]: *** [CMakeFiles/CubicSDR.dir/build.make:323: CMakeFiles/CubicSDR.dir/src/modules/modem/analog/ModemAM.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:104: CMakeFiles/CubicSDR.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
Error making: cubicsdr

Any fixes?

Dantali0n, 2019-09-17:
You need to update liquiddsp. 

```
cd wherever_liquiddsp_is
git pull origin master
./bootstrap.sh
CFLAGS="-march=native -O3" ./configure --enable-fftoverride
make -j4
sudo make install
sudo ldconfig
```

masterplayer30, 2019-09-17:
Failed!

vsonnier, 2019-09-17:
Oh no, not again ! Duplicate of #746, #749, #751, #752 please inform yourself there.

closed 2019-09-17 by vsonnier