cjcliffe/CubicSDR #757: Cannot open SoftRock device

n7ihq, 2019-09-22:
I am running CubicSDR version 0.2.5 on Debian 10. I tried to setup a SoftRock device using the following parameters:

SoapySDR Device Options > Run-time Settings
Rig Control: SoftRock Si570 AVR-USB
Rig Serial Rate: 19200 baud
Rig Serial Port: /dev/softrock

In the CubicSDR console log, I got the error "[ERROR] Rig failed to init.". The error occurs for any Rig Serial Rate value. I used the Hamlib rigctl command to test the USB interface of my SoftRock board. I got a similar error:

$ rigctl --model=2509 --rig-file=/dev/softrock --serial-speed 19200 --verbose
rig_open: error = IO error

The rig_open error was eliminated by removing the serial-speed option:

$ rigctl --model=2509 --rig-file=/dev/softrock --verbose
Opened rig model 2509, 'Si570 AVR-USB'

The SoftRock USB interface uses the "USB control transfer" mode and the Rig Serial Rate parameter is not required.

g0oan, 2020-01-15:
Hi,

I'm having a similar issue but while using a FA-SDR kit. It's similar to SoftRock in that it's a Si570 device and also uses the HamLib USB Kit backend to access the serial port on the SDR.

When I try to access the SDR directly from CubicSDR the 'Enable Rig' menu won't set no matter what settings I choose for Serial Rate, Control Port etc.

I have however gotten it to work using `HamLib's TCP daemon. On my Mac the command is

rigctld -m 2515 -r /USB -T 127.0.0.1 -t 4532

the rigctld command instructs the TCP Daemon, 2515 is the model number for my rig a Funkamateur FA-SDR, /USB is the port, on MacOS there is no need for any further port information and the remainder of the command tells the daemon to listen on TCP port 4532.

I then select HamLIb NET rigctl as the rig Model in CubicSDR's Rig Control menu and set 127.0.0.1:4532 as the rig Control Port. See the above rigctld command. 

Using this setup it all works, CubicSDR will tune and control the rig using the HamLib TCP daemon. Obviously that in itself confirms that the basic components are all working.

I suspect there is an issue somewhere with the settings on the Rig Control menu when CubicSDR is trying to access HamLib's USB kit backend directly. Would like to get it working if at all possible as it would save having to startup the TCP daemon whenever I want to use CubicSDR.

Hope this helps.

Kind regards,

Sean.

cjcliffe, 2020-06-17:
Hmm "rig_open error was eliminated by removing the serial-speed option" -- I'm starting to think that hamlib via code might behave the same as the command line; perhaps omitting the baud rate will also fix it there.  Right now the serial options are set whether the rig identifies itself as serial mode or not...