cjcliffe/CubicSDR #758: Mac compiled from source with Retina?

treaves, 2019-09-25:
The release version of the app runs & displays correctly.  I've just built from source, and it does not.  Every panel that isn't a standard widget only fills the left 50% of the area.  So for example the waterfall displays the entire waterfall, but, it only fills the left 50% of the area allocated to it.

Also the text is very small.  The buttons on the left hand side that display FM ,etcetera are unreadable.

This seems like may a Retina issue?  I've followed the instructions and compiled all dependencies from scratch.

Anything obvious for me to look at?

Thanks.

Magalex2x14, 2019-09-25:
I also use the version that I compile on my mac from sources. At the moment (I don’t know if this relates to the use of a retina display), I am experiencing a similar problem in symptoms, but they arise only after trying to scale the window with the mouse. After restarting the application, everything returns to normal. I am not competent in this matter, but if my memory serves me, some past display problems were related to wxWidgets - I think it’s worth looking at what they have with issues and updates...

treaves, 2019-09-26:
I've confirmed that on my non-Retina display, this displays correctly.  So it is something to do with the settings for Retina displays.  I'm not sure why the 'prod' app displays correctly, as any configuration file should be in the repo.

I'm not failure with wxWidgets so do not know where to look to check this.  I do remember needing to update my OpenGL based Qt apps when the HiDPI screens came on the market.

Magalex2x14, 2019-09-26:
@treaves It looks like not only we (mac users) are experiencing problems with  interface scaling on the HiDPI displays. Look at #727 
In addition, the wxWidgets team in [dev roadmap](https://trac.wxwidgets.org/wiki/Roadmap) promises version 3.1.3 in October (without specifying changes), but in the upcoming 3.2 they have a point about better support for high DPI displays...

Magalex2x14, 2019-09-26:
Somewhat interesting about this in the wxWidgets repo:
[Fix OpenGL samples when using HiDPI displays](https://github.com/wxWidgets/wxWidgets/pull/1470)
[Enable Retina (HiDPI) support on OSX wxGLCanvas](https://github.com/wxWidgets/wxWidgets/pull/1485)

treaves, 2019-09-26:
@Magalex2x14 good catches.  Unfortunately those two branches haven't yet been merged; I wonder why that is...  

Once they are I can try build from their master.

ra1nb0w, 2019-10-01:
Tried with that without success.

Magalex2x14, 2019-10-09:
After upgrading to MacOS Catalina I no longer have any problems with CubicSDR interface on the HiDPI display...

ra1nb0w, 2019-10-09:
I did a rebuild of CubicSDR and the latest wxwidgets-3.2 (through macports) and the problem is still there. What is your setup?

Magalex2x14, 2019-10-09:
wxWidgets 3.1.2 (release version, compiled from source)
CubicSDR 0.2.6a (latest master source)
I didn’t take any manipulations, but when I launched CubicSDR (builded on 10.14.6) after updating to Catalina 10.15 yesterday, I found that all the problems were gone...

ra1nb0w, 2019-10-09:
it happened a while ago during a beta of 10.15 but then the problem comes back.

treaves, 2019-10-09:
They're not gone.  The issue is not the OS; it's how wxWidgets handles HiDPI screens.  It requires a fix in wxWidgets.

ra1nb0w, 2019-10-09:
Sure. I only said that I saw a change during one of macOS 10.15 betas

ra1nb0w, 2019-10-29:
wxWidgets 3.1.3 seems doesn't resolv the problem.

Magalex2x14, 2019-11-07:
Builded today with wxWidgets 3.1.3 - the scaling issue is back, yes. And remained after reverting to 3.1.2 😭

Magalex2x14, 2019-11-08:
@ra1nb0w, thanks for this info! Now it became clear what to do and where to look...

With the changes below, I solved the problem of the quarter size of the OpenGL elements, but I still could not figure out how to correctly handle the font size. While I set it to 2x in CubicSDR options (screenshot), this setting does not affect all interface elements... I think it's worth calling someone who well knows CubicSDR source structure. @vsonnier, please, look at this changes. I think, they resolve HiDPI problems on Linux also.

Built on OS X 10.15.1 from latest master sources with wxWidgets 3.1.3.

[image]

```
From c3cccf7df12d260937dc7927e483855193fae644 Mon Sep 17 00:00:00 2001
From: Aleksey Makarenko <magalex28@gmail.com>
Date: Fri, 8 Nov 2019 23:08:39 +0300
Subject: [PATCH] proper OpenGL elements HiDPI handling

---
 src/ui/UITestCanvas.cpp           | 2 +-
 src/visual/GainCanvas.cpp         | 2 +-
 src/visual/MeterCanvas.cpp        | 2 +-
 src/visual/ModeSelectorCanvas.cpp | 2 +-
 src/visual/ScopeCanvas.cpp        | 2 +-
 src/visual/SpectrumCanvas.cpp     | 2 +-
 src/visual/TuningCanvas.cpp       | 2 +-
 src/visual/WaterfallCanvas.cpp    | 2 +-
 8 files changed, 8 insertions(+), 8 deletions(-)

diff --git a/src/ui/UITestCanvas.cpp b/src/ui/UITestCanvas.cpp
index 037258e..254272f 100644
--- a/src/ui/UITestCanvas.cpp
+++ b/src/ui/UITestCanvas.cpp
@@ -39,7 +39,7 @@ UITestCanvas::~UITestCanvas() {
 
 void UITestCanvas::OnPaint(wxPaintEvent& WXUNUSED(event)) {
   //  wxPaintDC dc(this);
-    const wxSize ClientSize = GetClientSize();
+    const wxSize ClientSize = GetClientSize() * GetContentScaleFactor();
     
     glContext->SetCurrent(*this);
     initGLExtensions();
diff --git a/src/visual/GainCanvas.cpp b/src/visual/GainCanvas.cpp
index eb7ac4f..ef30599 100644
--- a/src/visual/GainCanvas.cpp
+++ b/src/visual/GainCanvas.cpp
@@ -52,7 +52,7 @@ GainCanvas::~GainCanvas() {
 
 void GainCanvas::OnPaint(wxPaintEvent& WXUNUSED(event)) {
   //  wxPaintDC dc(this);
-    const wxSize ClientSize = GetClientSize();
+    const wxSize ClientSize = GetClientSize() * GetContentScaleFactor();
 
     glContext->SetCurrent(*this);
     initGLExtensions();
diff --git a/src/visual/MeterCanvas.cpp b/src/visual/MeterCanvas.cpp
index 63d67fc..840ebb2 100644
--- a/src/visual/MeterCanvas.cpp
+++ b/src/visual/MeterCanvas.cpp
@@ -83,7 +83,7 @@ void MeterCanvas::setShowUserInput(bool showUserInput) {
 
 void MeterCanvas::OnPaint(wxPaintEvent& WXUNUSED(event)) {
  //   wxPaintDC dc(this);
-    const wxSize ClientSize = GetClientSize();
+    const wxSize ClientSize = GetClientSize() * GetContentScaleFactor();
 
     glContext->SetCurrent(*this);
     initGLExtensions();
diff --git a/src/visual/ModeSelectorCanvas.cpp b/src/visual/ModeSelectorCanvas.cpp
index 16e4ff0..9044f2f 100644
--- a/src/visual/ModeSelectorCanvas.cpp
+++ b/src/visual/ModeSelectorCanvas.cpp
@@ -52,7 +52,7 @@ int ModeSelectorCanvas::getHoveredSelection() {
 
 void ModeSelectorCanvas::OnPaint(wxPaintEvent& WXUNUSED(event)) {
    // wxPaintDC dc(this);
-    const wxSize ClientSize = GetClientSize();
+    const wxSize ClientSize = GetClientSize() * GetContentScaleFactor();
 
     glContext->SetCurrent(*this);
     initGLExtensions();
diff --git a/src/visual/ScopeCanvas.cpp b/src/visual/ScopeCanvas.cpp
index 805cc9d..1e2bf6d 100644
--- a/src/visual/ScopeCanvas.cpp
+++ b/src/visual/ScopeCanvas.cpp
@@ -102,7 +102,7 @@ bool ScopeCanvas::getShowDb() {
 
 void ScopeCanvas::OnPaint(wxPaintEvent& WXUNUSED(event)) {
   //  wxPaintDC dc(this);
-    const wxSize ClientSize = GetClientSize();
+    const wxSize ClientSize = GetClientSize() * GetContentScaleFactor();
     
     ScopeRenderDataPtr avData;
     while (inputData->try_pop(avData)) {
diff --git a/src/visual/SpectrumCanvas.cpp b/src/visual/SpectrumCanvas.cpp
index ef886e0..25b1ed5 100644
--- a/src/visual/SpectrumCanvas.cpp
+++ b/src/visual/SpectrumCanvas.cpp
@@ -52,7 +52,7 @@ SpectrumCanvas::~SpectrumCanvas() {
 
 void SpectrumCanvas::OnPaint(wxPaintEvent& WXUNUSED(event)) {
   //  wxPaintDC dc(this);
-    const wxSize ClientSize = GetClientSize();
+    const wxSize ClientSize = GetClientSize() * GetContentScaleFactor();
     
     SpectrumVisualDataPtr vData;
     if (visualDataQueue->try_pop(vData)) {
diff --git a/src/visual/TuningCanvas.cpp b/src/visual/TuningCanvas.cpp
index 1fef0fc..c53a1bf 100644
--- a/src/visual/TuningCanvas.cpp
+++ b/src/visual/TuningCanvas.cpp
@@ -85,7 +85,7 @@ void TuningCanvas::setHalfBand(bool hb) {
 
 void TuningCanvas::OnPaint(wxPaintEvent& WXUNUSED(event)) {
  //   wxPaintDC dc(this);
-    const wxSize ClientSize = GetClientSize();
+    const wxSize ClientSize = GetClientSize() * GetContentScaleFactor();
     
     glContext->SetCurrent(*this);
     initGLExtensions();
diff --git a/src/visual/WaterfallCanvas.cpp b/src/visual/WaterfallCanvas.cpp
index a77ecb0..ad8c70c 100644
--- a/src/visual/WaterfallCanvas.cpp
+++ b/src/visual/WaterfallCanvas.cpp
@@ -130,7 +130,7 @@ void WaterfallCanvas::OnPaint(wxPaintEvent& WXUNUSED(event)) {
     std::lock_guard < std::mutex > lock(tex_update);
 //    wxPaintDC dc(this);
     
-    const wxSize ClientSize = GetClientSize();
+    const wxSize ClientSize = GetClientSize() * GetContentScaleFactor();
     long double currentZoom = zoom;
     
     if (mouseZoom != 1) {
-- 
2.20.1
```

ra1nb0w, 2019-11-12:
Works fine for the OpenGL widgets. You should send a PR with this patch.
Font: I also did some tests but without find the right way except mimic `fontScale` item.
Attach the font question to the PR.

Anyway, Thanks @Magalex2x14

ra1nb0w, 2020-04-03:
please, close this issue.