cjcliffe/CubicSDR #760: Center Frequency Scroll, Doesn't Tune

ithinkimongithub, 2019-10-10:
Using a mouse-wheel to scroll center frequency, changes adjusts the scale properly, and the center frequency increments, but the radio tuning remains at the old frequency.

e.g., tune 94.3 and then scroll only the 1's MHz place of center frequency, and listen to same station now on 95.3 Mhz. Appears to be only a case with 1's MHz place.
[image]
[image]

ithinkimongithub, 2019-10-10:
What this also means that I can return to 194.3MHz and still pick up 94.3MHz.
[image]

vsonnier, 2019-12-27:
Using a mouse-wheel to scroll center frequency, changes adjusts the scale properly, and the center frequency increments, but the radio tuning remains at the old frequency.

That is the general idea of SDR: the Center frequency is the tuning frequency, but any nimber of demodulators (VFOs) can be created in the range [Center - bandwidth / 2 ; Center + bandwidth / 2].
Once created, those demods have no reason to move from their "virtual" frequency and remain listenable if they are in the [Center - bandwidth / 2 ; Center + bandwidth / 2]. 
Once they are out of band, they become inactive and silent, of course.

You can see Cubic and SDRs that way: you can create any number of VFOs anywhere in you device reception full range (ex. 24MHz - 1500MHz for and RTL-SDR, 10KHz-2GHz for an RSP..) and they become live once your observation window, as large as your bandwidth, pass over them. That is How you explore the RF spectrum.    
 
Please inform further here #756 to find out how Cubic works. Sorry for having not much time to explain further.

closed 2019-12-27 by vsonnier