cjcliffe/CubicSDR #762: Segmentation fault with libX11.so.6.3.0 RSPduo

olivalle, 2019-10-16:
Launching CubicSDR the following error apperas :
No protocol specified
Unable to init server: Could not connect: Connection refused
17:36:16: Error: Unable to initialize GTK+, is DISPLAY set properly?

On dmesg : 
CubicSDR[2290]: segfault at 40 ip 00007f59518d94bb sp 00007ffc9ce66eb0 error 4 in libX11.so.6.3.0[7f59518b2000+8a000]

It seems a redundant problem with libX11.so.6.3.0

Dantali0n, 2019-10-17:
Could you specify more information such as:
Version of CubicSDR
Operating system
Distribution of OS + version
Machine architecture x86, armv8 etc
Desktop environment (KDE, Gnome, i3)

This seems related to your desktop environment as it can' t initialize GTK+ and asks if the $DISPLAY environment variable is correctly set.

vsonnier, 2019-12-27:
Issue seen countless times with not-the-proper wxWidgets (MUST be 3.1.3) or bad OpenGL setting.
Thanks @Dantali0n for trying to help, but since there is no feedback I'm closing it.

vsonnier, 2019-12-27:
Besides, the Duo support is not ready, see https://github.com/pothosware/SoapySDRPlay/issues/62 for information.

closed 2019-12-27 by vsonnier