cjcliffe/CubicSDR #766: Can't select audio output device under Ubuntu Linux

dslotter, 2019-11-01:
I see where I should be able to select audio devices under the Modem, but as you can see in this attached screen capture, I have four audio out devices, but only "PulseAudio" shows up in CubicSDR's GUI. This is under Ubuntu Linux 16.04. Does CubicSDR v0.2.5 under Linux not have the capability to send to a particular audio output device? I read the documentation and couldn't find any help on this. Please advise, and thanks.

[image]

righthalfplane, 2019-11-19:
In Ubuntu 16.04, you select the small speaker in the upper right hand corner of the screen to get to "Sound Settings".  In the sound dialog you can select the output device.

righthalfplane, 2019-11-19:
CubicSDR does not give you any options on Ubuntu. Raspbian gives you options, but ignores your selection.

tvjon, 2020-01-14:
I see there's an audio reference in issue 780, but I'll post here as doubtless folk read several similar issues.

I'm using raspbian on a RPi4, & have built CubicSDR now, but of course no audio, just plenty of queue full statements :(

I see there's no ALSA libraries referenced at all in CubicSDR's executable, so hard to see how alsa could work directly. I notice there's a couple of pulse libraries installed in raspbian Buster, but that's it.

So:
$ sudo apt install pulseaudio

$ alsamixer 

now shows the sound card as pulseaudio.

If you have Speakers on your HDMI monitor you should now get audio when running CubicSDR.

I don't have speakers on my current HDMI monitor, so need sound on RPi's output jack.

$ sudo apt install pavucontrol

This will install PulseAudio Volume Control into raspbian's  Raspberry Sound & Video sub-menu, where you can control volume. It also shows inputs & outputs. Next to divert audio to headphones.

Inside 

 ~/.config/pulse
 
make a file called:
 
default.pa

Edit that to contain:

.include /etc/pulse/default.pa
load-module module-switch-on-connect

Save the file then:

$ pulseaudio -k && pulseaudio --start

Now I can select headphones as the target destination for CubicSDR.

HTH

dreua, 2024-07-17:
With pipewire on Fedora one can just use the "Helvum" patchbay as a quick workaround: Just connect the outputs of "CubicSDR" with your output device. Not sure why this was necessary but I'm happy that it works now, maybe it helps someone else.

cjcliffe, 2026-11-15:
Latest RtAudio update on master should fix this -- RtAudio update enumerates individual devices now.

closed 2026-11-15 by cjcliffe