cjcliffe/CubicSDR #768: Won’t load on Mac OS Mojave

whcassing, 2019-11-07:
i have tried for several days to load the CubicSDR dmg file onto my iMac with Mojave.  Every attempt has ended with only 17k loading, even after several hours.

Dantali0n, 2019-11-07:
In order for us to be able to help you, we will need more information.

- Were do you download the dmg from with which url?
- Which version of the CubicSDR dmg are you downloading?
- What exactly do you mean with 'only 17k loading', does the download not work or are you unable to open the dmg image?
- If the download isen't working with which browser are you trying to download it and have you tried another browser?

whcassing, 2019-11-08:
I have attached two screen shots, one from Google and another from Firefox.  I have also tried Safari.  I tried to download from the GitHub software site, https://github.com/cjcliffe/CubicSDR/releases <https://github.com/cjcliffe/CubicSDR/releases>.  I am using an iMac running Mojave.

Dantali0n, 2019-11-09:
> I have attached two screen shots, one from Google and another from Firefox. I have also tried Safari. I tried to download from the GitHub software site, https://github.com/cjcliffe/CubicSDR/releases. I am using an iMac running Mojave.

Unfortunately, attaching images via email to reply to Github comments does not work. As a result we can not see the images you have attached.

@whcassing Could you upload the screenshots as attachment using the Github website? you can upload them by dragging & dropping them into the comment box.

whcassing, 2019-11-11:
Here are the images I sent in my last email.  Please note that the downloads stopped early.

Dantali0n, 2019-11-11:
> Here are the images I sent in my last email. Please note that the downloads stopped early.

I am sorry the images aren't sent, as again you replied via email and it is not possible to send images via email to Github.

Dantali0n, 2019-12-29:
@vsonnier should we close this one to? I do not think the author will come back to post the pictures and even than it seems like a local browser issue on the authors end. I can download and open all release files just fine.

vsonnier, 2019-12-29:
Thanks @Dantali0n for the report, let's close it then.

closed 2019-12-29 by vsonnier