cjcliffe/CubicSDR #771: Bookmarks in version 0.2.5 and suggestions for improving cubicSDR

ah1102, 2019-11-29:
Hello! Unfortunately, I noticed that you began to devote less time to such a wonderful program as CubicSDR (((

- I was faced with the fact that in version 0.2.5 the frequencies in the bookmarks cannot be assigned a text name, it simply is not saved.

- I also noticed that in the program there is no setting for disabling AGC by sound frequency and its manual adjustment. This causes discomfort when using a squelch and switching modes of modulation. Almost always, the AGC restores the sound signal level for a very long time.

- I would also like to see in the program the audio output to the Network, since, for example, the DSD + application supports receiving audio stream over the network. Gqrx has this feature. I would like to see in your application.

- I want to say the same about managing the program through the Network. This makes it easy to connect applications like Gpredict to track satellites and compensate for Doppler shift. The Gqrx program has such an opportunity.

- It would be great if a function appeared in your program to display the name of the bookmarks and the range in the frequency spectrum window, as is implemented in Gqrx and SDR #.

P.S. I use mac os x and I think your application is the best on this platform. I would like to see the further development of CubicSDR!

cjcliffe, 2020-06-17:
@ah1102 thanks for the submission; I'm glad you find the application useful.  Apologies that I haven't been doing much open-source development for awhile now but @vsonnier and others have done an excellent job in my absence.

- Hopefully the bookmark names are something simple; I see another issue #792 has reported the same.
- Sounds like the audio AGC should be getting reset when modes are switched but probably isn't.
- Is there a standard protocol for Network Audio that you can direct me to?
- I believe you can currently use gpredict by hooking it up to hamlib's net rigctl (see #308) but direct control would be nice to be on par with other SDR applications
- Thanks for adding the bookmark suggestion #797 

I finally have a newer Mac here than can run the latest MacOS and XCode so I should be able to handle some of the more recent bugs and pull requests.

ah1102, 2020-06-18:
- In the current source code, someone has already fixed this issue #792, so we are waiting for release 0.2.6 on all platforms.

- Audio AGC recovers a very long time when switching from NBFM to AM or from AM to SSB

- GQRX uses the UDP protocol, and the DSD decoder is successfully connected to the program, in the settings of which it was specified to listen to the network address
[image]

ah1102, 2020-06-18:
Congratulations on your purchase! Mac for the programmer is the best that can be)