cjcliffe/CubicSDR #775: New release? (58086c17f: ampmodem parameters have changed)

df7cb, 2019-12-25:
Hi,
I'm updating the liquid-dsp package in Debian, and the ampmodem change (https://github.com/jgaeddert/liquid-dsp/commit/22ccf5fb34a5a4253d984b2f1f81c67b7aa6eab7) makes the current CubicSDR release 0.2.5 incompatible. I see this is already fixed in master (58086c17f).
It would be nice to see a new CubicSDR release including this change.
Thanks!

vsonnier, 2019-12-27:
Duplicate of countless others, like #755. I leave it open though, as a friendly ping to @cjcliffe.

df7cb, 2019-12-27:
Thanks for the cross-links. I was looking for other issues, but couldn't find any with "release".

Fwiw, Debian is still on wxgtk 3.0 so even 0.2.5 won't build unless the 3.1 patch is reverted (which we do at the moment, https://salsa.debian.org/debian-hamradio-team/cubicsdr/blob/master/debian/patches/revert-migration-to-wx3.1), but putting up a new release seems to make sense in any case.

vsonnier, 2023-01-13:
Forgot to close this antique as duplicate.

closed 2023-01-13 by vsonnier